mletenay/home-assistant-goodwe-inverter #119: Entities lost in HA 2022.4

zoesn89, 2022-04-07:
Hey guys,

first of all: Thank you so much for your great work!!! :wink:

Since updating HA to 2022.4, I have no longer all entities in my Goodwe Integration. In 2022.3 I had about 150 entities. Now, I only have 10?! I removed and installed the integration, but I had no success -.-

Can anybody help?

Best
Felix

DewanPieterse, 2022-04-07:
Hello

I have the same issue with ES model on firmware 2121F. 

Used to have 62 entities, now only 5. Important ones are lost like battery power and on-grid export.

Thanks
Dewan

liebefuroscar, 2022-04-07:
+1. Not working with Goodwe EM5000 and HA 2022.4.0

mletenay, 2022-04-07:
HA developers deprecated and removed some parts of API which the extension was using.
The v0.9.9.1 should work with HA 2022.4.0 fine again.
Enjoy !

closed 2022-04-07 by mletenay

dani2dani, 2022-04-07:
Can confirm v0.9.9.1 works with HA 2022.4.0 fine. Thank you!

liebefuroscar, 2022-04-08:
Working for me too. Thanks for the quick solution. :)

DewanPieterse, 2022-04-08:
Thanks! That was quick!