mletenay/home-assistant-goodwe-inverter #120: Limit battery charge

doppelrittberger, 2022-04-12:
Hi
I live in Germany and we have to limit feed into grid to 70%. In most cases the battery will be charged in the morning and around 1pm when sun generates most power battery is full and inverter needs to limit.
Is there a way to limit the battery loading speed or soc? So I could use home assistant to prevent battery from loading in the morning and instead load later.
Alternative would be to limit the loading to max ampere. Is this possible? I have ET plus series.
Big thanks for any help

mletenay, 2022-04-12:
Hmm, I actually have the same problem. 
In normal modes, the battery charge currents seem to be managed by the battery BMS.
But with the Eco mode (and especially these new synthetic Eco Charge and Eco Discharge) modes, there is this "Eco mode power" (in % of inverter nominal power?).
I haven't tried that yet, but I guess switching in the morning to Eco Charge and set some charge limit could solve the problem.
(And in the afternoon switch to Eco Discharge or General mode)

mletenay, 2022-11-28:
Limiting charge/discharge parameter can be achieved by calling get_parameter/set_parameter services with appropriate parameter name (e.g. battery_charge_limit or battery_discharge_limit).

closed 2022-11-28 by mletenay