mletenay/home-assistant-goodwe-inverter #123: Integration fails to load in HA core 2022.5 - invalid entity ID

sramepa1, 2022-05-21:
Hello. It seems that this integration has not been updated yet to the breaking change of https://github.com/home-assistant/core/pull/66835 , so when loading it under HA core 2022.5, it logs `Invalid entity ID: .goodwe_inverter`

I tried to fix it myself and submit a PR but I failed to find the right place where to fix the `entity_id` to include a prefix as I am not familiar enough with HA API.

mletenay, 2022-05-23:
That's odd, I do not observe any problems in HA 2022.5.5.
And the error seems to be related to the "inverter" entity which was removed quite a while ago.
Could it be it is just some obsolete/disabled/unavailable entity entry which is causing problems ?

sramepa1, 2022-05-23:
Thanks for looking into this. I have tried removing the Goodwe integration and adding it again, freshly configuring it with the inverter IP address again, and also the "Reload integration" button in HA. Neither helped. Is there a better way to perform cleanup?

mletenay, 2022-05-23:
Just a guess - try to look at the (Settings>Devices>>>Entities), show disabled & hidden entities and try to remove any goodwe disabled entity.

sramepa1, 2022-05-23:
I have since downgraded HA to 2022.4 where everything works fine. When I try upgrading next time, I'll make sure to check that tab as well. If 0.9.9.1 works normally under HA 2022.5 for you, feel free to close this as "cannot reproduce". Thank you for your time.

closed 2022-08-09 by mletenay