mletenay/home-assistant-goodwe-inverter #125: issue starting up with HA core-2022.6.6

ziporah, 2022-06-22:
During startup this message is thrown:

```
2022-06-22 10:08:10 INFO (SyncWorker_2) [homeassistant.util.package] Attempting install of goodwe==0.2.17
2022-06-22 10:08:11 WARNING (MainThread) [homeassistant.components.statistics.sensor] The configuration parameter 'state_characteristic' will become mandatory in a future release of the statistics integration. Please add 'state_characteristic: mean' to the configuration of sensor 'goodwe_pv_stats' to keep the current behavior. Read the documentation for further details: https://www.home-assistant.io/integrations/statistics/
2022-06-22 10:08:34 INFO (MainThread) [homeassistant.setup] Setting up goodwe
```

Using main version from github repo and latest HA core-2022.6.6

mletenay, 2022-06-24:
This is a bit weird, since I do not see that warning and I even do not think the integration creates 'goodwe_pv_stats' sensor at all.
Isn't it your custom sensor ?

ziporah, 2022-06-24:
oh crap 🤦‍♂️ 
```
sensor:
  - platform: statistics
    entity_id: sensor.goodwe_ppv
    name: goodwe_pv_stats
    max_age:
      minutes: 5
```
I was totally confused with HA refusing to startup and failed to load the goodwe the sensors. I had to upgrade my custom addon already to latest github version due to the recent core changes and was thinking this warning was left behind.