mletenay/home-assistant-goodwe-inverter #127: 5048ES stopped working after latest firmware update

tinuva, 2022-07-08:
My 5048ES inverter was updated from 2121 to 2424 and now the integration fails to load. 

Goodwe mentioned this when doing the update:
> At the moment you are on 212110/  we now have 242416/.

Logs that I see when HA start:
```
2022-07-08 09:59:36 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 10.0.0.59.
2022-07-08 09:59:36 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2022-07-08 09:59:37 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2022-07-08 09:59:38 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3234323447475735303438442d45532331302020202020202020202020202039353034384553553230325730303037333630303431302d30343032352d32340e3431302d30323033342d31360211de
```

No errors or anything else, only in HA Integration page I see this:
[image]

This is HACS version 0.9.9.1 and I also tried master, same thing.

tinuva, 2022-07-08:
Ah I see it was fixed with https://github.com/marcelblijleven/goodwe/issues/19

So for now I manually updated my manifest.json to
```
  "requirements": ["goodwe==0.2.18"],
```

dakota, 2022-07-22:
Thanks for the fix! My Goodwe ES inverter got updated last night and I was getting the same error. Works again, hopefully there'll be a official release soon.

DarthV02, 2022-08-01:
hi @tinuva, @dakota, i understand you had the integration already installed when it started failing, didn't you? Hence you updated the manifest.json... But, any idea if the bug / update is been already sorted / released? I have tried to install it through the HA integration UI but no luck... all the time i get the failure: 

> Logger: aiohttp.server
> Source: components/goodwe/config_flow.py:36
> First occurred: July 30, 2022 at 9:44:00 PM (3 occurrences)
> Last logged: 8:05:00 PM
> 
> Error handling request
> Traceback (most recent call last):
> 
> File "/usr/local/lib/python3.10/site-packages/goodwe/es.py", line 160, in read_device_info
> self.arm_sw_version = int(self.arm_version[4], base=16)
> ValueError: invalid literal for int() with base 16: 'G'

I also read in some other post that due to the inverter firmware update, the base=16 shall be changed to base=32 in file es.py... but since I can't install the application i can't do any amendments on the code. 

Any ideas?  
Thanks in advance :)

tinuva, 2022-08-01:
> hi @tinuva, @dakota, i understand you had the integration already installed when it started failing, didn't you? Hence you updated the manifest.json... But, any idea if the bug / update is been [...]

Only works if you install the custom integration. Eg. through HACS. Then it is easy to find the manifest.json file since it is in a place that is close to your configuration.yaml file. If you use the official goodwe integration, then the code is more difficult to find.

So in HACS, install it. In fact, if you over ride the version and install from master, then you will get the fix. However a release have not been published yet by @mletenay that includes fix.

So either install a released version and edit the file. or install the master branch in HACS and you will have the fix.

DarthV02, 2022-08-01:
hi @tinuva, 

Thanks for the quick response! My apologies i'll keep on asking... :) when you refer to the HACS version, just to confirm, you mean the GoodWe Inverter (experimental), don't you??:

https://github.com/mletenay/home-assistant-goodwe-inverter

Once installed i should run the inverter_scan.py, shouldn't i? then i get the next error (no module named 'goodwe'):

```
File "\\192.168.1.135\config\custom_components\goodwe\select.py", line 4, in <module>
    from goodwe import Inverter, InverterError
ModuleNotFoundError: No module named 'goodwe'
PS Microsoft.PowerShell.Core\FileSystem::\\192.168.1.135\config>
```

i've cheked in line 4 and others whether it was a problem just related to the name of my inverter (which i don't know if it has a name anyway). 

More tips?

tinuva, 2022-08-02:
@DarthV02 that is the correct url yes.

Don't worry about running inverter_scan.py - thats only when you want to debug things. For now you only need to install the integration.

So your HA folder should look like this:
configuration.yaml
custom_components/goodwe
custom_components/goodwe/manifest.json
custom_components/goodwe/const.py
etc.

Then when you add the goodwe integration in the UI, it will use the custom_component instead of the built-in one.

After adding the files as above, you need to restart HA before adding the integration.

DarthV02, 2022-08-02:
Hi @tinuva , I understand now :) 
System is up and running. Many thanks for your quick support!

mletenay, 2022-08-09:
Fixed in v0.9.9.2

closed 2022-08-09 by mletenay