mletenay/home-assistant-goodwe-inverter #129: Help understanding of the 60 sensors

oziee, 2022-07-10:
Hi.. great to see that we can show goodie inverters in the HA system :) Im trying to work out which sensors to use to make templates to show in the energy dashboard

I have a GW5000S-BP (DEBUG:goodwe:Detected ES/EM/BP inverter GW5000S-BP)

Looking at the SEMS and PV Master app and HA sensors side by side, I must be missing some of what each sensor is actually referring to or how its being calculated.

Since there is no PV inputs on the SBP inverter, there still seems to be data reflected in the sensors that there is #128 

But trying to work out the battery info there seems to be 2 sensors that are nearly identical - total power & battery power
[image]

what is total power vs battery power?

the main sensors.. the only one that seems right and correct is the battery percent.. all the other values do not seem to align to anything
[image]

[image]

House consumption and PV power do not match 

which sensors are needed to the energy dashboard?? power to grid, power from grid, battery in out?

XanderF, 2023-04-07:
I'm curious about the numbers as well. I have a GW3000S-BP and the numbers in HA just make no sense at all. The app and SEMS are spot on. It would be great if somehow those numbers good end up in HA.

I use sensor.battery_power to calculate power buy/sell and use that in the energy panel. That's an approximation, but it's pretty close (see main page -> https://github.com/mletenay/home-assistant-goodwe-inverter)

JoshC1994, 2023-08-14:
It would be great to have a library added to the wiki of how each of the sensors is determined what each value is will help a lot of us newbies and good refence material.