mletenay/home-assistant-goodwe-inverter #137: Goodwe ES5048 - trying to select eco mode - charging/discharging

DennisDav774, 2022-09-12:
Hi

When selecting eco charge mode / eco discharge mode the following appear in the logs:

Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/goodwe/select.py:99
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 16:09:48 (1 occurrences)
Last logged: 16:09:48

[139951800269072] bad operand type for abs(): 'str'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1738, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1775, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 931, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 54, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 99, in async_select_option
    await self._inverter.set_operation_mode(operation_mode, self._eco_mode_power)
  File "/usr/local/lib/python3.10/site-packages/goodwe/es.py", line 234, in set_operation_mode
    await self.write_setting('eco_mode_1', EcoMode("1", 0, "").encode_discharge(eco_mode_power))
  File "/usr/local/lib/python3.10/site-packages/goodwe/sensor.py", line 251, in encode_discharge
    return bytes.fromhex("0000173b{:04x}ff7f".format(abs(eco_mode_power)))
TypeError: bad operand type for abs(): 'str'

svecpetr, 2022-11-13:
Could I have a question? Where is possible to change operation mode to Eco charge / discharge mode ... I see only Eco mode. Thanks for reply.

mletenay, 2023-02-05:
I assume these problems should be fixed in v0.9.9.7 now

closed 2023-02-05 by mletenay