mletenay/home-assistant-goodwe-inverter #138: Battery Discharge

BossLurker, 2022-09-14:
Hi

Firstly thank you for a great integration!! 

I have one issue that since the last update I can no longer force the battery to discharge into the grid... 
I can set the eco mode power and select eco discharge but as long as there is grid power it wont discharge. 

There are no errors in the logs that I can find. 

Thank you 

Mat

Jkarlik28, 2022-09-17:
I have i same problem. In general mode cannot battery to discharge into the grid.

BossLurker, 2022-09-22:
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/goodwe/select.py:105
Integration: GoodWe Inverter (documentation, issues)
First occurred: 10:45:13 PM (1 occurrences)
Last logged: 10:45:13 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 105, in update_eco_mode_power
    self._eco_mode_power = int(float(event.data.get("new_state").state))
ValueError: could not convert string to float: 'unavailable'

BossLurker, 2022-09-22:
Not sure if this helps?

Traceback (most recent call last):
File "/config/custom_components/goodwe/select.py", line 105, in update_eco_mode_power
self._eco_mode_power = int(float(event.data.get("new_state").state))
ValueError: could not convert string to float: 'unavailable'
2022-09-22 23:28:16.891 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.50.217.
2022-09-22 23:28:16.892 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2022-09-22 23:28:16.900 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2022-09-22 23:28:17.216 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3234323447475735303438442d45532331302020202020202020202020202039353034384553553139333030343232333630303431302d30343032352d3234203431302d30323033342d31360211d3
2022-09-22 23:28:17.219 DEBUG (MainThread) [goodwe] Connected to inverter GW5048D-ES, S/N:95048ESU19300422.
2022-09-22 23:28:17.223 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2022-09-22 23:28:17.736 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0000000000000000000002020001005000dd000300000025000064006461020000010996000802cb13850109960021003f1385010200f00000000000044ac700004bad011a017e0005d978ffea0100200202f400010000fc85010800700000160916171c1300000000000000000000000000000000000000000000000000002e5e000028ba020000000000001553
2022-09-22 23:28:17.739 DEBUG (MainThread) [custom_components.goodwe] Finished fetching GoodWe data in 0.520 seconds (success: True)
2022-09-22 23:28:17.749 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2022-09-22 23:28:18.082 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000173b00000000000000000001000100020001000a004a02140062002601bd0015000f000000000000006402140172001e0069196400000000000002260000000700030000000000010172013001b300140000000c08e5
2022-09-22 23:28:18.089 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2022-09-22 23:28:18.414 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000173b00000000000000000001000100020001000a004a02140062002601bd0015000f000000000000006402140172001e0069196400000000000002260000000700030000000000010172013001b300140000000c08e5
2022-09-22 23:28:18.420 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2022-09-22 23:28:18.822 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000173b00000000000000000001000100020001000a004a02140062002601bd0015000f000000000000006402140172001e0069196400000000000002260000000700030000000000010172013001b300140000000c08e5
2022-09-22 23:28:18.826 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2022-09-22 23:28:19.228 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000173b00000000000000000001000100020001000a004a02140062002601bd0015000f000000000000006402140172001e0069196400000000000002260000000700030000000000010172013001b300140000000c08e5
2022-09-22 23:28:19.240 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b0701040000173b0003ff7f0463
2022-09-22 23:28:19.341 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b901060300
2022-09-22 23:28:19.344 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b07050430003000006400000358
2022-09-22 23:28:19.439 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b901060300
2022-09-22 23:28:19.443 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b0709043000300000640000035c
2022-09-22 23:28:19.539 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b901060300
2022-09-22 23:28:19.543 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b070d0430003000006400000360
2022-09-22 23:28:19.650 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b901060300
2022-09-22 23:28:19.654 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f032c050000173b0302c7
2022-09-22 23:28:19.747 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003ac010602f4
2022-09-22 23:28:19.751 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f032d050000173b0302c8
2022-09-22 23:28:19.846 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003ad010602f5
2022-09-22 23:28:19.850 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f033601000278
2022-09-22 23:28:20.162 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003b6010602fe
2022-09-22 23:28:20.166 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f03590103029e
2022-09-22 23:28:20.516 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003d901060321
2022-09-22 23:28:20.520 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f031d00025e
2022-09-22 23:28:20.773 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc0039d010602e5
2022-09-22 23:28:22.002 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2022-09-22 23:28:22.476 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0000000000000000000002020001005000dd000300000025000064006461020000010992000002a91384010992001a006b1384010100f00000000000044ac700004bad011a017e0005d978ffb403002002023e00010000fcae010800700000160916171c1800000000000000000000000000000000000000000000000000002e5e000028ba020000000000001487
2022-09-22 23:28:22.480 DEBUG (MainThread) [custom_components.goodwe] Finished fetching GoodWe data in 0.479 seconds (success: True)
2022-09-22 23:28:27.002 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
[... cut ...]
2022-09-22 23:45:16.697 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0000000000000000000001fd0001005000dc000300000025000064006461020000010996000802121377010996001a000b1377010200ef0000000000044ac700004bad011a017f0005d979ffea01002002023600010000fcdc010800700000160916172d1200000000000000000000000000000000000000000000000000002e5e000028ba0200000000000014e6
2022-09-22 23:45:16.699 DEBUG (MainThread) [custom_components.goodwe] Finished fetching GoodWe data in 0.699 seconds (success: True)

svecpetr, 2022-11-20:
> I have i same problem. In general mode cannot battery to discharge into the grid.

General mode cannot be never used for charge batter from grid or discharge batery into grid.
Only mode that can do it is "Eco mode" means not ecological but economical mode...

mletenay, 2023-02-22:
The EcoMode on ES inverters was reimplemented in v0.9.9.9, I guess it should also address this issue.

closed 2023-02-22 by mletenay