mletenay/home-assistant-goodwe-inverter #142: Update to official HA Goodwe integration

rnovacek, 2022-11-19:
First of all, thanks for great integration.

Are there any plans to update the official HA Goodwe integration, esp. with the new "Eco charge mode" and "Eco discharge mode"?

I recently saw a facebook discussion (in [Czech HomeAssistant users group](https://www.facebook.com/groups/2232679967058877/posts/3381580002168862/)) where people got confused about not having those two new states while others have them.

Or should I ask on Home Assistant side in the issues there?

svecpetr, 2022-11-21:
It would be greate if HA Goodwe integration were upgraded to latest version in HACS

mletenay, 2022-11-28:
Well, the Eco charge/discharge are still considered experimental, there are problems with it on ES inverters, so it will remain in HACS only for a while.

rnovacek, 2022-11-28:
Cool, thanks for letting us know. I'll switch to HACS for now.

closed 2023-12-21 by mletenay