mletenay/home-assistant-goodwe-inverter #143: Eco charge mode

svecpetr, 2022-11-20:
Changing mode to "Eco charge mode" setup a enable timer, that has this settings
- from 0:00 to 23:59 ... OK
- with power of 0% ... BAD ... need to be set to 100% for user manually value (charging with 0% do nothing, because 0% = 0W)
- battery mode is set to Discharge ... BAD ... Eco charging mode should set mode to CHARGE not discharge :/

janchlebek, 2022-11-20:
This is not a bug - you need to specify what power you want to use for discharging via a service. Not everybody wants to charge with 100% or any other predefined value. The inverter unfortunately tends to forget what the last value was so it is best to simply set this value whenever setting Eco modes in a sequence or using script.

Also there are two modes - whichever you select it does exactly what it is supposed to do - Eco charge mode charges battery, Eco discharge mode discharges. With both you have to set the value in percent of inverter total power (i.e. for ET-10k setting 20 will charge or discharge with 2kW of power.

svecpetr, 2022-11-21:
Sorry and thanks. Solved... It's a shame there isn't a proper manual

closed 2022-11-21 by svecpetr