mletenay/home-assistant-goodwe-inverter #144: Eco discharge mode

svecpetr, 2022-11-20:
After changing inventer mode to Eco discharge mode is created / enabled timer with predefined power of 0% ... That is bug. Discharging at 0% of power of inventer means discharging at 0W ... and that means do nothing.

It is needed to setup this timer with power of 100% or with already user predefined value like 70%

janchlebek, 2022-11-20:
This is not a bug - you need to specify what power you want to use for discharging via a service. Not everybody wants to charge with 100% or any other predefined value. The inverter unfortunately tends to forget what the last value was so it is best to simply set this value whenever setting Eco modes in a sequence or using script - for example: 
`    sequence:
      - device_id: ID_of_inverter
        domain: select
        entity_id: select.inverter_operation_mode
        type: select_option
        option: Eco charge mode
      - device_id: ID_of_inverter
        domain: number
        entity_id: number.eco_mode_power
        type: set_value
        value: XY`

janchlebek, 2022-11-20:
> it it percent of power of inventer (for example 10KW inventer has 100% as 10KW and it did not say nothing about ampers of discharging, because you need to know the battery voltage

The inverter will take care of this and if maximum battery current reported by BMS  is lower than the power set in the operation mode it will reduce it. Always the lower value of the two will be used.

svecpetr, 2022-11-21:
Yes, you are right.

Sorry and thanks. Solved... It's a shame there isn't a proper manual

closed 2022-11-21 by svecpetr