mletenay/home-assistant-goodwe-inverter #147: 2 integrations

rhoracek, 2022-12-09:
Hello, please, add the option of integrating 2 inverters, when adding the second inverter, the integration ends with an error. Thank you.

rhoracek, 2022-12-10:
[host_2022-12-10T11-23-02.797Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10199955/host_2022-12-10T11-23-02.797Z.log)
[home-assistant_2022-12-10T11-22-52.092Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10199957/home-assistant_2022-12-10T11-22-52.092Z.log)
[home-assistant_2022-12-10T11-17-08.294Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10199958/home-assistant_2022-12-10T11-17-08.294Z.log)
[home-assistant_2022-12-10T11-15-00.820Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10199959/home-assistant_2022-12-10T11-15-00.820Z.log)

closed 2022-12-10 by rhoracek

mletenay, 2022-12-13:
There are no error logs related to goodwe, try to increase log level for the integration.