mletenay/home-assistant-goodwe-inverter #148: Login Failed

lowieh, 2022-12-15:
When i try to add this integration i have to fill in the ip-address of the converter. I know for sure the ip is the right one.
i type 192.168.1.150 and wait for about 30 seconds and then failed to connect. 
What iam doing wrong.

What is the goal for me. To see all the information from the inverter like temperature, voltage and many more in my home assistant application.

Rincovanderbeek, 2022-12-24:
I got the same with a GW3000-NS

mletenay, 2022-12-27:
GW3000-NS inverters are known to contain older ARM firmware version, so it is likely that is the reason you cannot connect to it.
See note in https://www.home-assistant.io/integrations/goodwe/
Safest way to check if the inverter is supported is to try the official mobile app (PvMaster or SolarGo). If that work, this integration should as well. If not, then you need firmware upgrade.

Rincovanderbeek, 2022-12-28:
Thanks for your kind reply.
I had a call with Godwe support, unfortunately this inverter could not be updated.
They told me the ARM  processor of this conveter is not capable of handling PvMaster oid.
So I'll return the convetre and buy a newer one.

closed 2023-02-05 by mletenay