mletenay/home-assistant-goodwe-inverter #150: Missing Configuration.

ZS1FJK, 2022-12-20:
Hello

I have a goodwe GW10K-ET with Firmware: 04029-08-S11 (02041-19-S00). 
On the APP I have a setting called fast charge.  It is found in the advanced settings => Battery settings. I am able to set the percentage that I can charge the battery from mains power. I wish this option was always enabled on the inverter but once the inverter has charged itself to the setpoint it turns fast charge off. I am hoping that I can automate this using home assistant. This is to make sure the battery has a minimum charge at all times to avoid long times at low voltages and also when the power goes out we have enough in the battery at night for the bear essentials. We currently have no power for up to 8 hours a day at the moment. 

Here are the photos of the app
[image]
[image]

Thank you for all the work done on this add on device for HASS I have automated so many aspects of my home to save and consume power when there is sun and when there is no power. We have reduced our power footprint by 50 % because of this add on alone.

Klaus3d3, 2023-01-24:
I believe the same task can be achieved via setting to eco mode. But thats just a guess.

mletenay, 2023-02-05:
The "fast charging" inverter feature should be reachable in v0.9.9.7 now.
It is not exposed in UI anywhere, but you should be able to apply it by calling _set_parameter_ service - there are now parameters _fast_charging_ (0-1), _fast_charging_soc_ (10-100), _fast_charging_power_ (10-100)

closed 2023-02-05 by mletenay

TomEkk33, 2023-02-16:
Is there a way to add  "fast charging" option in a near future to the main HA? Thanks.

I cannot easily switch to HACS due to lack of "Add-ons" - my HW uses "Add-ons" (like MQTT, others)

ZS1FJK, 2023-02-21:
Thank you.
I will give this a try this weekend. Thank you

TomEkk33, 2023-02-26:
I checked this with ET inverter and it works OK! Thanks a lot.
[mletenay](https://github.com/mletenay) - It would be great to have "fast charging" in official HA.

drydenmike, 2023-10-16:
Hey. Just to let you know I appreciate the work you're doing here.
Having said this, I have a ES model and fast charging is not working with any of the parameters mentioned above. Do you think this might be different depending on software version? Is there a way to find out which parameters I can use in mine? Thanks

drydenmike, 2023-10-23:
@mletenay any updates on my last question? Sorry to bother you. I'd appreciate a nudge in the right direction.

traveler6666, 2024-04-15:
Hi,

I really love this featrue. However, shortly after I set up the service, I can't set the SOC anymore. I think there was an update. 
I am getting an unpexted error message and the protocoll says: goodwe.exceptions.RequestRejectedException: ILLEGAL DATA VALUE if the value is e.g. 81 or input_number.fast_charging_soc.

The code is:
  - service: goodwe.set_parameter
    data:
      device_id: b9cb944ba3a1382249238c4a2523b698
      parameter: fast_charging_soc
      value: states("input_number.fast_charging_soc")

Is there an error in the code or a bug in the integration, was it worked before?