mletenay/home-assistant-goodwe-inverter #151: Charging/discharging plan reset after changing the mode via HA

tauceti2, 2022-12-22:
Hi,

First of all thanks for a great integration of GoodWe! I have started to use modes, however when I set charging/discharging plan via mobile app and then change the mode via HA (e.g. from eco to general), all the plan entries are disabled. When I activate them again they remain active till another mode change via HA. I have GoodWe 8K-ET.

Thanks for any help in advance.

Michal

TomEkk33, 2023-01-09:
Hi,

I'd like to connect to this issue and confirm that observe the same behavior on ET inverter. Going from General Mode (or other mode) to  ECO disables ECO schedules. The hours,values remain as previously set, but they are inactive. 
Goodwe allows to setup 4 different plans in ECO. So in ECO mode should be an option in HM to direct which 
"Battery power groupX" should be activated in ECO (where X=1,2,3,4).

The only solution for now is activation ECO plans manually in "PV Master" app each time when switching ECO out/in.

Please help how to fix this. Thanks.

BR,
Tom.

janchlebek, 2023-01-09:
Not a solution from my end - but how about simply ditching PV Master and doing everything via HA (create corresponding schedules in HA)?

TomEkk33, 2023-01-09:
> Not a solution from my end - but how about simply ditching PV Master and doing everything via HA (create corresponding schedules in HA)?

Well, HA doesn't have access (or at least I cannot find them) to some parameters which are visible inside Goodwe ET inverters. 
They are available with PV Master only. Beside of this I'd like to stay with Goodwe original settings to have all 
in sync in case of HA issues.

I'll try to use [elRadix](https://github.com/elRadix) solution. Thanks.

janchlebek, 2023-01-09:
> Well, HA doesn't have access (or at least I cannot find them) to some parameters which are visible inside Goodwe ET inverters. They are available with PV Master only. Beside of this I'd like to stay [...]

Just out of curiosity - which parameters you mean? The HA integration exposes almost everything. 

Moreover even if you use HA as me or elRadix suggest you can still use PV Master to read data - by ditching PV Master I meant to not use it to control charging, you can still use it to access whatever information you want but leave charging control on some HA automation. I have a very similar one in place as elRadix and it works great. If HA fails, you can again set your schedules in PVM.

TomEkk33, 2023-01-09:
> Just out of curiosity - which parameters you mean? The HA integration exposes almost everything.

"Battery power groupX"  in ECO (where X=1,2,3,4) is listed in PV Master. I don't know which parameter in HA is responsible for these.
Do you know which entity to look (select.inverter_operation_mode and then which option?) Thanks.

[image]

elRadix, 2023-01-09:
maybe delete them and let HA create a new one ...

TomEkk33, 2023-01-09:
> maybe delete them and let HA create a new one ...

Sorry, I don't understand. What should be deleted in HA: whole Goodwee device and add it again in HA?

elRadix, 2023-01-09:
sorry I meant in the pv master app, the different eco modes

TomEkk33, 2023-01-09:
> sorry I meant in the pv master app, the different eco modes
ECO mode without these "different eco modes" does not make any use to me.

Those modes are disabled by default when HA sets General Mode and then again back to the ECO mode.
The question is how to enable them again using HA "Eco mode"? Thanks.

janchlebek, 2023-01-10:
> "Battery power groupX" in ECO (where X=1,2,3,4) is listed in PV Master. I don't know which parameter in HA is responsible for these. Do you know which entity to look (select.inverter_operation_mode and then which option?) Thanks.

But isn't this "battery group" just the schedule? Or you have more battery modules connected that you control separately somehow? I am not at home now so have no access with PV Master...

mletenay, 2023-02-22:
I have re-implemented the EcoMode behavior in v0.9.9.9, can you verify it works now properly ?

TomEkk33, 2023-02-26:
I made some basic checking of ECO charge, discharge modes with ET inverter. Seem to work OK using "Battery power Group1".
[image]

I noticed that "Battery power Groups 2,3,4" are overwritten with some default values. 
It would be useful if you could do the deactivation of these groups only - leaving their other values unchanged (time, day of week, type).

TomEkk33, 2023-02-26:
The Peak Shaving Mode (https://github.com/mletenay/home-assistant-goodwe-inverter/issues/145)  is quite similar to "ECO charge mode". The PSM requires also few parameters - currently they are not set when switching to PSM.
[image]

mletenay, 2023-03-12:
I have changed the Eco Mode Charge/Discharge switching in the library v0.2.28 (will get to integration release v.0.9.9.10).
It should no longer clear the schedules 2-4 and will just turn them off.

closed 2023-03-12 by mletenay

filipkrystl, 2023-06-16:
It would be great if you could reenable disabled schedules after going back to normal "Eco mode" from either "Eco Mode Charge/Discharge" or from "General". Or set in configuration those schedules that should be enabled when going to "Eco mode" 

My scenario is that I have some schedules to charge/discharge battery in PV master for Eco mode and now mode change in HA will currently disable them and there is no way from HA to enable them again. I have to go to PV master a enable them manually. Thanks