mletenay/home-assistant-goodwe-inverter #152: Interface often not available

robgiom, 2022-12-31:
The interface wirth GoodWe Inverter is not work properly since I'm reporting continuing no availability for several seconds and new  connections

[image]

Could you help me in order to understand the reason why it happen?

Many thanks in advance for your help

TheAnT1979, 2023-01-11:
i have the same issue

Kdonkers, 2023-02-13:
Bump, same issue

mletenay, 2023-02-13:
These connection issues are most likely wifi related.
Increase log level for goodwe module and provide some logs.

Kdonkers, 2023-02-13:
I have direct connection via LAN module, so this seems incorrect.

mletenay, 2023-02-13:
Well, actually there are reports of users that LAN is actually less reliable than Wifi.
Cannot neither confirm or deny that personally, but the fact is the network infrastructure is major source of the problems.
If you are using HACS version, you can try to adjust the timeout and retry parameters, for some users it helped.

closed 2023-12-21 by mletenay