mletenay/home-assistant-goodwe-inverter #154: voltage vs. frequency device class

Paja-git, 2023-01-26:
Since HA version 2023.1. I've noticed the startup warnings like: _"WARNING__ (MainThread) [homeassistant.components.sensor] Entity sensor.goodwe_backup_f1 (<class 'custom_components.goodwe.sensor.InverterSensor'>) is using native unit of measurement 'Hz' which is not a valid unit for the device class ('voltage') it is using; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration _author."_
Warnings are common for all freq. sensors.

mletenay, 2023-01-26:
Should be fixed in v0.9.9.6

closed 2023-01-26 by mletenay

Paja-git, 2023-01-27:
Looks fine, thank you.
Pj

From: mletenay
Sent: Thursday, January 26, 2023 11:44 PM
To: mletenay/home-assistant-goodwe-inverter
Cc: Paja-git; Author
Subject: Re: [mletenay/home-assistant-goodwe-inverter] voltage vs. frequencydevice class (Issue #154)

Should be fixed in v0.9.9.6