mletenay/home-assistant-goodwe-inverter #158: Eco mode power is unavailable

sotoniak, 2023-02-15:
In version 0.9.9.6 the Eco charge power setting is unavailable. Furthermore, Eco charge mode and Eco discharge mode cannot be enabled.
(goodwe gw10k-et)
[image]

mletenay, 2023-02-16:
The individual items are enabled upon request to inverter (to check if it is supported by the inverter model/fw).
Since the UDP is unreliable, sometimes it happens that even after 3 retries the proper response is not received.
Could you restart the HA to ensure it was not case like this ?

mletenay, 2023-02-16:
Actually, there seems to be really a bug there ! Going to fix it.

mletenay, 2023-02-16:
Fixed in v0.9.9.8

closed 2023-02-16 by mletenay