mletenay/home-assistant-goodwe-inverter #160: Connection Lost for Goodwe GW5000D-NS

carlodie, 2023-02-18:
Hi dears, since 4 days ago I lost the connections, sensors and reports from my Goodwe Inverter, seems it's related to the new Home Assistant version **2023.2.5** Please if can support to me to solve this issues, this is the information and logs:

**Device Information**

> GW5000D-NS
> By GoodWe
> Firmware: 51.51.32 (None)

**Logs:**

> 2023-02-18 09:33:01.225 WARNING (MainThread) [async_upnp_client.profiles.dlna] Only InstanceID 0 is supported
> 2023-02-18 09:33:01.405 WARNING (MainThread) [async_upnp_client.profiles.dlna] Only InstanceID 0 is supported
> 2023-02-18 09:33:05.279 WARNING (MainThread) [homeassistant.config_entries] Config entry 'GoodWe' for goodwe integration not ready yet: ('No valid response received even after 10 retries', 1); Retrying in background
> 2023-02-18 09:33:08.265 ERROR (MainThread) [homeassistant.components.esphome.dashboard] Error requesting ESPHome Dashboard data: Cannot connect to host 127.0.0.1:64048 ssl:default [Connect call failed ('127.0.0.1', 64048)]
> 2023-02-18 09:34:52.868 DEBUG (MainThread) [goodwe] Connecting to DT family inverter at 192.168.0.66.
> 2023-02-18 09:34:52.869 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
> 2023-02-18 09:34:52.874 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409
> 2023-02-18 09:34:53.876 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:53.876 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #1/10
> 2023-02-18 09:34:54.878 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:54.879 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #2/10
> 2023-02-18 09:34:55.880 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:55.881 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #3/10
> 2023-02-18 09:34:56.883 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:56.884 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #4/10
> 2023-02-18 09:34:57.886 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:57.887 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #5/10
> 2023-02-18 09:34:58.890 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:58.891 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #6/10
> 2023-02-18 09:34:59.893 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:34:59.893 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #7/10
> 2023-02-18 09:35:00.895 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:35:00.896 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #8/10
> 2023-02-18 09:35:01.899 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:35:01.899 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #9/10
> 2023-02-18 09:35:02.902 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:35:02.903 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #10/10
> 2023-02-18 09:35:03.907 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request 7f03753100280409 failed.
> 2023-02-18 09:36:24.397 DEBUG (MainThread) [goodwe] Connecting to DT family inverter at 192.168.0.66.
> 2023-02-18 09:36:24.397 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
> 2023-02-18 09:36:24.402 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409
> 2023-02-18 09:36:25.405 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:25.405 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #1/10
> 2023-02-18 09:36:26.408 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:26.409 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #2/10
> 2023-02-18 09:36:27.412 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:27.412 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #3/10
> 2023-02-18 09:36:28.415 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:28.416 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #4/10
> 2023-02-18 09:36:29.418 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:29.418 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #5/10
> 2023-02-18 09:36:30.420 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:30.420 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #6/10
> 2023-02-18 09:36:31.422 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:31.422 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #7/10
> 2023-02-18 09:36:32.425 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:32.426 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #8/10
> 2023-02-18 09:36:33.429 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:33.429 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #9/10
> 2023-02-18 09:36:34.434 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:36:34.434 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #10/10
> 2023-02-18 09:36:35.438 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request 7f03753100280409 failed.
> 2023-02-18 09:37:55.739 DEBUG (MainThread) [goodwe] Connecting to DT family inverter at 192.168.0.66.
> 2023-02-18 09:37:55.739 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
> 2023-02-18 09:37:55.742 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409
> 2023-02-18 09:37:56.746 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:37:56.746 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #1/10
> 2023-02-18 09:37:57.749 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:37:57.749 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #2/10
> 2023-02-18 09:37:58.752 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:37:58.752 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #3/10
> 2023-02-18 09:37:59.756 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:37:59.756 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #4/10
> 2023-02-18 09:38:00.758 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:38:00.759 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #5/10
> 2023-02-18 09:38:01.763 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:38:01.763 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #6/10
> 2023-02-18 09:38:02.765 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:38:02.766 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #7/10
> 2023-02-18 09:38:03.768 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
[... cut ...]
> 2023-02-18 09:38:04.771 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:38:04.772 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #9/10
> 2023-02-18 09:38:05.775 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to 7f03753100280409 in time (1s).
> 2023-02-18 09:38:05.776 DEBUG (MainThread) [goodwe.protocol] Sending: 7f03753100280409 - retry #10/10
> 2023-02-18 09:38:06.777 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request 7f03753100280409 failed.

mletenay, 2023-02-18:
Hmm, the inverter does not seem to respond at al.
And there were no changes to DT/D-NS inverters recently, I think it is just a coincidence and has nothing to do with new AH version.
I don't know how to help, just try several restarts and re-check the connection inverter (incl. SolarGo app)

carlodie, 2023-02-18:
> Hmm, the inverter does not seem to respond at al. And there were no changes to DT/D-NS inverters recently, I think it is just a coincidence and has nothing to do with new AH version. I don't know how to help, just try several restarts and re-check the connection inverter (incl. SolarGo app)

Thanks, SolarGo and SEMS Portal works fine... Also no errors or alarms in the Inverter... The connection lost suddendly after HA 2023.2.5, not only for Goodwe also in my "Tuya" IoT integration.

Maybe this logs can give you a clue, thanks in advance again...

[home-assistant_goodwe_2023-02-18T16-04-35.646Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10774737/home-assistant_goodwe_2023-02-18T16-04-35.646Z.log)
[home-assistant_tuya_2023-02-18T16-25-39.426Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/10774738/home-assistant_tuya_2023-02-18T16-25-39.426Z.log)

carlodie, 2023-02-18:
Hi [mletenay](https://github.com/mletenay) after review the logs of my router I found the IP of the inverter was changed (seems the IP of Goodwe was set at the begin of integration as DHCP).
After that I set the IP as static (as was set at the beginning by the router), this action restore the connections and all the sensors and metrics works again :) 
Please if you know how I can change the IP for Goodwe integration, I prefer asign a new IP in a reserved segment of my router.

mletenay, 2023-02-20:
Just remove the integration and add it back again - it will ask the IP during setup.
All the sensors and their history should be preserved (but do backup just to be sure ;-).

closed 2023-02-20 by mletenay