mletenay/home-assistant-goodwe-inverter #161: Parameters cannot be changed

Plumpop1, 2023-02-18:
On the GoodWe ES 5048 inverter, after the update to 0.9.9.8, the parameters general, eco charge, eco discharge cannot be changed.

bubakbrno, 2023-02-19:
The same problem on GW3600S-BP . Since 0.9.9.8 works perfect.

Plumpop1, 2023-02-19:
> The same problem on GW3600S-BP . Since 0.9.9.8 works perfect.

Ahoj. Jak to myslíš že od 0.9.9.8 funguje perfektně? Mě to funguje perfektně od verze 0.9.9.7

DewanPieterse, 2023-02-20:
Same for me.

`Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/goodwe/select.py:131 
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22)) 
First occurred: 12:18:29 (15 occurrences) 
Last logged: 12:41:20

[140477978537616] ES.set_operation_mode() takes from 2 to 3 positional arguments but 4 were given
[140477949340368] Error handling message: Unknown error (unknown_error) from 102.39.129.185 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.3 Safari/605.1.15)
[140477949340368] ES.set_operation_mode() takes from 2 to 3 positional arguments but 4 were given
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1787, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1824, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 968, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 720, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 56, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 131, in async_select_option
    await self._inverter.set_operation_mode(
TypeError: ES.set_operation_mode() takes from 2 to 3 positional arguments but 4 were given`

mletenay, 2023-02-20:
Seems like I broke the code for ES inverters.
Please revert back to prev version for the time being, fix (with proper Eco V2 for ES) will come soon.

mletenay, 2023-02-22:
v0.9.9.9 was just released and should fix the issues.

closed 2023-02-22 by mletenay