mletenay/home-assistant-goodwe-inverter #162: ET series Eco Mode problem

LubosKovar, 2023-02-21:
After last update my GW10-ET does not charging in eco_charge mode adn stay in Standby mode. Eco_SOC has not effect. FW inverter 1023. Before work normal.

mletenay, 2023-02-22:
Can you change it in PvMaster ?
The trouble is, that since my inverter was upgraded to fw 1023, the PvMaster crashes when trying to set Eco Mode there, so I can't properly test it.

abernas, 2023-02-22:
I have an inverter 8et firmware: 04029-06-S11 / 02041-17-S00, which has been the same for several months and the problem occurred only after the update to the latest version of the add-on. When I switch eco mode charge in Ha Error: Failed to Call the Select/Select_option service. Operation Not Supported

LubosKovar, 2023-02-22:
> Can you change it in PvMaster ? The trouble is, that since my inverter was upgraded to fw 1023, the PvMaster crashes when trying to set Eco Mode there, so I can't properly test it.

Yes, I can change it in PVMaster and switch enter to ecomode normaly. I have iPhone with iOS16 and last version of PVMaster. Inverter is GW10-ET (without +) and last FW 1023

mletenay, 2023-02-22:
v0.9.9.9 was just released and should fix the issues.

closed 2023-02-22 by mletenay