mletenay/home-assistant-goodwe-inverter #163: Inverter IP address change

Delorien84, 2023-02-21:
Hello, I have had problem with inverter that IP address change due to the router upgrade (not have setup static lease). After IP address change the HA integration stop working. According to log the integration tries to connect to the old IP address again and again. I have to change IP address directly in `config/.storage/core.config_entries` file to new one.

Is there any "better" way how to specify inverter IP address?  Or way how to "clear" the settings? 

Thank you for the answer.

janchlebek, 2023-02-21:
I suggest to make a reservation for the (old) IP on your router and then restart the inverter. It should then get the old IP. 

Alternatively remove the integration from HA and add it again using new IP - this however will not fix the problem permanently as the IP might change again in the future.

Delorien84, 2023-02-21:
Yes, I already did it. But it was kind surprise to me as there is no host/ip address setting, nor anything in documentation. So one would assume that the integration will automatically discover the inverter.

I just want to ask if it's possible to expose inverter ip address as configurable variable or do some discovery retry if old ip address does not response.

Thank you

rkrejcir, 2023-04-28:
+1 This feature would be great to have. I know you can set static IP in inverted settings, but when I do it then the inverter wifi led starts freaking out and SEMS portal is unavailable.

mletenay, 2024-05-16:
IP is now exposed via configuration UI -  v0.9.9.20

closed 2024-05-16 by mletenay