mletenay/home-assistant-goodwe-inverter #164: Inverter ES 5048

Plumpop1, 2023-02-22:
Hi Mletenay
After the update to v0.9.9.9, the modes will already switch and PV Master will also show the change. However, the inverter runs in the original set mode without any changes.
Inverter ES 5048 firmware 2121E

mletenay, 2023-02-24:
This seems to be a consequence of https://github.com/home-assistant/core/issues/81788
Can you check the "wrong" behavior applies to all modes, or only to some of them ?

tinuva, 2023-02-24:
> This seems to be a consequence of home-assistant/core#81788

Oh you should have answered that person, that on ES, if you don't reboot
after mode change, the change doesnt take affect. The reboot is required on
ES.

Plumpop1, 2023-02-25:
No reboot needed on ES.

tinuva, 2023-02-25:
Then why did you open this issue? Test it in oc master. Set the mode, then
when it ask for reboot say no. Watch new mode not take effect. Try again,
say yes for reboot see if it works.

Plumpop1, 2023-02-25:
Everything works as it should in PV Master. With the ES inverter, there is no need to restart when changing the mode. This is just a recommendation. In version v0.9.9.7, everything works without reboot. I opened this topic because v0.9.9.8 did not change modes at all and version v0.9.9.9 changes modes but there is no change in any mode. I additionally check everything via PV Master. What do you still have Tinuva with the reboots? No Goodwe inverter needs to be restarted. This is just a recommendation.

tinuva, 2023-02-25:
I don't know, on my 5048ES if I don't reboot, the mode change does nothing.

Kind regards,
David Bezuidenhout

Plumpop1, 2023-02-25:
I don't want to burden this thread about how Goodwe inverters work. I have installed about 30 of them and have 4 in operation at home. But if you control from PV Master, change the mode from normal mode to eco and exit the application without confirming the restart. Close the application hard. Then go back to the PV Master application and you will see that you have eco mode. The change takes about 10 seconds.

tinuva, 2023-02-25:
Yes it will show the mode change, but the actual operation doesnt change.
But you are right, let's stop arguing. I currently am forced to not use the
mode change because it doesn't work.

mletenay, 2023-02-25:
Guys, since I do not have ES inverter, I cannot fix it without knowing what to do.
You need to help me to identify the proper steps to change mode.
The best would be if you are able to checkout the library and test the change mode there, looking at previous versions of the code.
I can help guiding you, but someone with ES needs to do the stuff and report inverter behavior.

tinuva, 2023-02-25:
Can do. Could you post the lines I need to add to the check script to test
this on the udp discord channel or here?

Plumpop1, 2023-02-25:
Since I'm not a programmer, I don't know where to look for codes that change. But I can do remote access via public ip address.
Mletenay what approach should i prepare for you?

mletenay, 2023-02-27:
Guys, I just went through the code and individual versions and something does not add up.
The change removing those restarts was done on Nov 11 and was already included in v0.9.9.7 which you say works fine.
So it does not seem to be the source of the problems.
v0.9.9.8 was a bit unfortunate release, so it might not work well.
But there are no relevant changes between 0.9.9.7 and 0.9.9.9 as far as I can see.
Are you really sure 0.9.9.7 works fine (and 0.9.9.9 not) ?

Plumpop1, 2023-02-27:
Hello Mletenay. Yes, version 0.9.9.7 runs without a problem. Now I reinstalled version 0.9.9.9. The problem with this version is that it changes the working mode to eco charge and eco discharge, but the charging or discharging power does not change when you try to change it. At the same time, PV Master writes that the mode has been changed. Off grid mode works fine. I rolled back to version 0.9.9.7 and everything works as it should without a reboot.

Plumpop1, 2023-02-27:
> 

> Guys, I just went through the code and individual versions and something does not add up. The change removing those restarts was done on Nov 11 and was already included in v0.9.9.7 which you say works fine. So it does not seem to be the source of the problems. v0.9.9.8 was a bit unfortunate release, so it might not work well. But there are no relevant changes between 0.9.9.7 and 0.9.9.9 as far as I can see. Are you really sure 0.9.9.7 works fine (and 0.9.9.9 not) ?
> 
> I do not really have any explanation, just it is maybe not version related but only a coincidence and unreliable behavior ? Can you set the 'goodwe' debug level to DEBUG and check if there aren't some failed requests during modes changes ?

> Guys, I just went through the code and individual versions and something does not add up. The change removing those restarts was done on Nov 11 and was already included in v0.9.9.7 which you say works fine. So it does not seem to be the source of the problems. v0.9.9.8 was a bit unfortunate release, so it might not work well. But there are no relevant changes between 0.9.9.7 and 0.9.9.9 as far as I can see. Are you really sure 0.9.9.7 works fine (and 0.9.9.9 not) ?
> 
> I do not really have any explanation, just it is maybe not version related but only a coincidence and unreliable behavior ? Can you set the 'goodwe' debug level to DEBUG and check if there aren't some failed requests during modes changes ?

How do I set the debug level?

mletenay, 2023-02-28:
edit configuration.yaml and put or adjust logger section like this:
```
logger:
  default: warning
  logs:
    components.goodwe: debug
    goodwe: debug
```

mletenay, 2023-02-28:
@Plumpop1 I am a bit confused what is working and what is not.
The normal modes (general, backup) do work without restart properly.
The Eco mode (standard one) does work properly (just accepts what was configured in PvMaster before).
The Eco Charge/Discharge work only partially - the mode is selected but the power is not ?
What is the power in PvMaster then ? What about the max charge SoC ? Does it work or not ?
I need exact descriptions which modes and which parameters do not work and what it means they do not work - what are the values expected/actual.

Plumpop1, 2023-02-28:
> edit configuration.yaml and put or adjust logger section like this:
> 
> ```
> logger:
>   default: warning
>   logs:
>     components.goodwe: debug
>     goodwe: debug
> ```

I created a yaml

Plumpop1, 2023-02-28:
I understand. I'll try to describe it better.
1. Normal mode works properly. It doesn't even allow any other settings. PV Master displays ok
2. Off Grid Mode works properly. PV Master displays ok
3. Backup mode works correctly PV Master displays ok
4. Eco mode PV Master displays ok but when changing the power scale of eco mode power, no change in the power of the inverter is reflected
5. Eco charge mode PV Master displays ok but again no response when changing the power on the scale.
6. Eco discharge mode same result as point 5.
7. I have no way to test SOC charging if the inverter does not start charging in any mode.
8. Test settings in HAS: Eco Charge Mode, eco mode power 40%, not charging, PV Master shows charging and power 10%

[image]
[image]

mletenay, 2023-02-28:
I see now.
You have inverter with older firmware, so the EcoModeV2 (the one with max charge SoC) is not applicable.
So you have the EcoModeV1 and that one seems to completely ignore the "power" value and shows 10%.
I will investigate it, but meanwhile:
1) In PvMaster, try to switch mode to General, then to Eco and set there some power value.
2) Restart HA and check if the Eco Mode Power slider read the value (set in PvMaster) properly.
3) Then try to switch to General mode in HA and back to Eco Charge and verify it still does not work.

Plumpop1, 2023-02-28:
1. So after restarting HA it read the performance percentages fine.
2. OK in HA on general mode.
3. After switching to eco charge mode, it does not work.
I also wrote to Goodwe support about updating the firmware, but it can take up to 14 days
My point with the whole thing is to be able to use SOC charging
Thank you very much for your time.

tinuva, 2023-03-05:
ok so out of interest sake.

```
~/Downloads/goodwe/tests (git)-[master] % python3 inverter_check.py
2023-03-05 10:16:03,031 __init__(59) - DEBUG: Using selector: KqueueSelector
2023-03-05 10:16:03,032 connect(51) - DEBUG: Connecting to ET family inverter at 10.0.4.10.
2023-03-05 10:16:03,032 _ensure_lock(123) - DEBUG: Creating lock instance for current event loop.
2023-03-05 10:16:03,032 _send_request(66) - DEBUG: Sending: f70388b800213ac1
2023-03-05 10:16:03,346 datagram_received(49) - DEBUG: Received: aa55f70342000113ec0000003900350030003400380045005300550047005700350030003400180018008a000000003431302d30343032352d32343431302d30323033342d3030c99c
2023-03-05 10:16:03,346 connect(53) - DEBUG: Connected to inverter GW504, S/N:95048ESU.
Identified inverter
- Model: GW504
- SerialNr: 95048ESU
- Rated power: 5100
- A/C output type: 0
- Firmware: 410-04025-24
- ARM firmware: 410-02034-00
- Modbus version: 1
- DSP1 version: 24
- DSP2 version: 24
- DSP svn version: 138
- Arm version: 0
- ARM svn version: 0
```

On version 0.9.9.7 and 0.9.9.9 version, this is working for me without the reboot.

I believe the reboot was required when I was on a super old firmware. However now that both my normal firmware and arm firmware has been updated, I can confirm that the mode change from general to eco charge, works 100% without a reboot. I suspect the reboot was needed when I was on the very old arm firmware. So this is a very nice and welcome!

mletenay, 2023-03-05:
@tinuva Try the Beta/master version of the HACS integration.
The Eco V2 should be working there.

tinuva, 2023-03-05:
Oh brilliant!

Can confirm eco v2 works!

Many thanks @mletenay

robotz0mbie, 2023-03-19:
hi, I have the same problem

> 1. So after restarting HA it read the performance percentages fine.
>
>     2. OK in HA on general mode.
>
>     3. After switching to eco charge mode, it does not work.
>        I also wrote to Goodwe [...]

I have the same problem with version >0.9.9.7, with a 5048-ES. PV master shows my firmware version as 2222C, did the firmware update fix your issue? thanks :)

Plumpop1, 2023-03-19:
I'm still waiting for an update.
I currently have 1212114

robotz0mbie, 2023-03-19:
thanks! do we know which firmware on the ES 5048 supports eco V2/ setting ECO SOC via the plugin at all?
I didn't do much testing, will do a bit tomorrow.
but basically, I had upgraded to the new version that supports setting the SOC for ECO mode (great!!) but I had not tested it.
Today I needed to add some charge to the battery, tried (via HA, no ECO profiles set in PV master at all) setting ECO SOC to target, then enabling ECO charge mode (eco power at 100%), no charge. I didn't actually notice if the mode changed or not, just, that it didnt work. (v.0.9.9.9)
Restarting HA - the ECO mode SOC setting was retrieved (so I guess it was saved to the inverter?) but charge/discharge still didn't work.
So I rolled back to 0.9.9.8 - same issue.
rolled back to 0.9.9.7 (Eco charge/discharge worked previously with this version for me) = working again.
I also noticed the ECO mode power % is a bit strange, the power level did not scale as expected (ie. 10-20% setting was still putting in 40-50% power, and no change after about 70%), will test that again tho hmmm.

What would be helpful to log to confirm firmware is the issue? I will log changing modes with v.9.7, then update to v.9.9 and log setting the ECO SOC and changing modes?

Plumpop1, 2023-03-19:
For me, version 0.9.9.7 works without problems, when I put version 0.9.9.9 the eco modes do not work.

tinuva, 2023-03-19:
> thanks! do we know which firmware on the ES 5048 supports eco V2/ setting ECO SOC via the plugin at all? 

If your ARM firmware is new enough, then yes!

I tested eco v2 by setting the target SoC to 80% while battery was on ~70% at night, and battery was charged to 81% then stopped, on master branch, I think some time after 0.9.9.9?
[image]

robotz0mbie, 2023-03-19:
hmmm. SEMS app shows FW 6232315 for me. So I might ask goodwe to update ARM and main firmwares, hopefully they can do that...!
and if I change to the master branch, the mode change no longer works at all, an error pops up down the bottom showing something like "only 3/4 arguments given" ! strange?

robotz0mbie, 2023-03-24:
so, goodwe updated the ARM firmware today (I asked to update both, but, I think only the arm version has changed), now version 6232317.
I re-downloaded 0.9.9.9 and gave it a go, but still could not get ECO discharge mode working (didn't try charge, as battery at 100% right now hmm). 
here is the log, here, I first changed ECO power to 50%, then ECO SOC to 50%, then changed mode from general to Eco Discharge. However, no discharging happened hmmm. 

`2023-03-24 18:05:22.141 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:22.533 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d29000d020c3a00090202150002005000fa000c00640064000063006363010000010931001a000313870100670000026b13870102016c000000000000cf0a0000139300f1006300009d9c026a01003000000000010000ffe401000800000017031812051400000000000000000000000000000000000000000000000000004df500004c6f02000000000000130b
2023-03-24 18:05:27.142 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:27.592 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d16000c020c2600090202140002005000fa000600640064000063006363030000010929001a000113880100670000026513880102016b000000000000cf0a0000139300f1006300009d9c026d01003000000000010000fffb01000800000017031812051900000000000000000000000000000000000000000000000000004df500004c6f0200000000000012ee
2023-03-24 18:05:32.142 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:32.657 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d0a000b020bea00090202140002005000fa000000640064000063006363030000010924001a000613890100680000027c13890102016b000000000000cf0a0000139300f1006300009d9c025901003000000000010000ffb401000800000017031812051e00000000000000000000000000000000000000000000000000004df500004c6f020000000000001362
2023-03-24 18:05:37.142 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:37.538 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d20000b020bd600090202140002005000fa0000006400640000630063630300000109220019000013890100670000025313890102016c000000000000cf0a0000139300f1006300009d9c024c01003000000000010000ffe601000800000017031812052400000000000000000000000000000000000000000000000000004df500004c6f02000000000000135d
2023-03-24 18:05:42.122 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b0701080000173b0032ff7f0496
2023-03-24 18:05:42.189 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b90115030f
2023-03-24 18:05:42.196 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:42.711 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0cf2000b020bd600090202140002005000fa00000064006400006300636303000001092000190004138a01006700000265138a0102016b000000000000cf0a0000139300f1006300009d9c024d01003000000000010000ffd501000800000017031812052900000000000000000000000000000000000000000000000000004df500004c6f020000000000001438
2023-03-24 18:05:42.763 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b0705083000300000640000035c
2023-03-24 18:05:42.830 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b90115030f
2023-03-24 18:05:42.840 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b07090830003000006400000360
2023-03-24 18:05:42.907 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b90115030f
2023-03-24 18:05:42.914 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f02390b070d0830003000006400000364
2023-03-24 18:05:43.248 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc002b90115030f
2023-03-24 18:05:43.253 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f033601000278
2023-03-24 18:05:43.649 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003b6010602fe
2023-03-24 18:05:43.654 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f03590103029e
2023-03-24 18:05:43.972 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc003d901060321
2023-03-24 18:05:47.143 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:47.531 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d1a000b020bae00090202140002005000fa00000064006400006300636303000001091d0019000713890100670000027613890102016b000000000000cf0a0000139300f1006300009d9c024e01003000000000010000ffb301000800000017031812052e00000000000000000000000000000000000000000000000000004df500004c6f02000000000000132c
2023-03-24 18:05:52.141 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:52.592 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d02000b020bd600090202140002005000fa00000064006400006300636303000001091d001a000813860100670000027d13860102016b000000000000cf0a0000139300f1006300009d9c025901003000000000010000ffdc01000800000017031812053300000000000000000000000000000000000000000000000000004df500004c6f020000000000001378
2023-03-24 18:05:57.141 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2023-03-24 18:05:57.721 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d2d000b020bc2000a0202140002005000fa00000064006400006300636303000001091e001a000113870100670000026013870102016b000000000000cf0a0000139300f1006300009d9c026601003000000000010000000101000800000017031812053800000000000000000000000000000000000000000000000000004df500004c6f0200000000000011a7
`

I notice in PV master, when changing the mode in HA from general to Eco / Eco charge / Eco discharge, the mode changes to Economic mode, but yeah... thats it, nothing else. Any ideas? could still be a FW issue?

tinuva, 2023-03-24:
@robotz0mbie

Eco discharge seems to work fine for me, except it just stops using PV then use battery for loads. Probably affected by the export limit setting where mine is set to prevent export.

Try downloading the master version instead of a numbered version.

In addition, I always ask for firmware and arm updates separate.
1x ticket for firmware update
1x ticket for arm update

robotz0mbie, 2023-03-24:
> @robotz0mbie
>
> Eco discharge seems to work fine for me, except it just stops using PV then use battery for loads. Probably affected by the export limit setting where mine is set to prevent export. [...]

thanks, I will try that with the ticket!
when using the master version, I get the following error when trying to switch modes (popup at bottom of HA screen)

Failed to call service select/select_option. Aa55ReadCommand.__init__() takes 3 positional arguments but 4 were given.

Plumpop1, 2023-03-25:
Hello friends.
So after the firmware update of my ES inverter on version 1242417/0
Everything works perfectly. ECO charging, eco discharge, SOC Limit, and eco mode power. I'd like to share with you a mistake but I didn't find any. Thank you Mletenay for your work and your time.

robotz0mbie, 2023-03-25:
> Hello friends. So after the firmware update of my ES inverter on version 1242417/0 Everything works perfectly. ECO charging, eco discharge, SOC Limit, and eco mode power. I'd like to share with you a mistake but I didn't find any. Thank you Mletenay for your work and your time. Version v.0.9.9.9

awesome news :)

mletenay, 2023-03-26:
Is everything working fine on ES now ? 
Can I close the ticket, or is there still something to fix ?

Plumpop1, 2023-03-26:
Thank you all right to close.

closed 2023-03-26 by Plumpop1