mletenay/home-assistant-goodwe-inverter #165: Cannot find entity and integration gives ILLEGAL DATA ADDRESS

Diamantino-Op, 2023-02-24:
I'm trying to connect to my inverter, but it keeps giving me ILLEGAL DATA ADDRESS and the entity is not found,

this is the log:
2023-02-24 17:01:12.934 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.105.
2023-02-24 17:01:12.934 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-02-24 17:01:12.969 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-02-24 17:01:18.485 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70342000111f800000039003500300034003800450053004100470057003500300034001700170079000000003431302d30343032352d32333431302d30323033342d303058c7
2023-02-24 17:01:18.554 DEBUG (MainThread) [goodwe] Connected to inverter  G W 5 0 4, S/N: 9 5 0 4 8 E S A.
2023-02-24 17:01:18.657 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-02-24 17:01:18.718 WARNING (MainThread) [homeassistant.components.sensor] Platform rest not ready yet: Server disconnected without sending a response.; Retrying in background in 30 seconds
2023-02-24 17:01:18.785 WARNING (MainThread) [homeassistant.components.sensor] Platform rest not ready yet: Server disconnected without sending a response.; Retrying in background in 30 seconds
2023-02-24 17:01:19.570 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa170218110111061300030000002e05eb00040000003c00000000000000000000000000000000000000220863005e138bfffff7fa00000000000000000000000000000000000000000001fffff7fa00000000000000000000000008420003138b00010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011db0001015a000000001b0018fe000002210176000007f600030000000000010002000000000000001900000019000000020000000200020000000000000000006c006c0000004400440000000000000001000000000000000000000000000001000c000000000000002313
2023-02-24 17:01:20.099 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-02-24 17:01:20.524 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA ADDRESS.
2023-02-24 17:01:20.525 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f7830220c3
2023-02-24 17:01:20.544 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 1.990 seconds (success: False)
2023-02-24 17:01:20.545 WARNING (MainThread) [homeassistant.config_entries] Config entry 'GoodWe' for goodwe integration not ready yet: ILLEGAL DATA ADDRESS; Retrying in background
2023-02-24 17:01:23.839 WARNING (MainThread) [homeassistant.helpers.frame] Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for hacs using this method at custom_components/hacs/__init__.py, line 171: hass.config_entries.async_setup_platforms(
2023-02-24 17:01:24.144 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.105.
2023-02-24 17:01:24.145 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-02-24 17:01:24.193 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-02-24 17:01:24.551 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70342000111f800000039003500300034003800450053004100470057003500300034001700170079000000003431302d30343032352d32333431302d30323033342d303058c7
2023-02-24 17:01:24.556 DEBUG (MainThread) [goodwe] Connected to inverter  G W 5 0 4, S/N: 9 5 0 4 8 E S A.
2023-02-24 17:01:24.610 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-02-24 17:01:25.305 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa17021811011705db00030000002c05d500040000003b00000000000000000000000000000000000000220867005e138bfffff7f600000000000000000000000000000000000000000001fffff7f600000000000000000000000008460003138b00010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011e80001015a000000001b001901000002210175000007f000030000000000010002000000000000001900000019000000020000000200020000000000000000006c006c0000004400440000000000000001000000000000000000000000000001000c000000000000000fef
2023-02-24 17:01:25.316 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-02-24 17:01:25.472 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA ADDRESS.
2023-02-24 17:01:25.472 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f7830220c3
2023-02-24 17:01:25.479 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.922 seconds (success: False)
2023-02-24 17:01:35.794 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.105.
2023-02-24 17:01:35.795 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-02-24 17:01:35.798 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-02-24 17:01:36.125 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70342000111f800000039003500300034003800450053004100470057003500300034001700170079000000003431302d30343032352d32333431302d30323033342d303058c7
2023-02-24 17:01:36.128 DEBUG (MainThread) [goodwe] Connected to inverter  G W 5 0 4, S/N: 9 5 0 4 8 E S A.
2023-02-24 17:01:36.131 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-02-24 17:01:36.888 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa17021811012205a800030000002b05d100040000003b00000000000000000000000000000000000000220863005e138bfffff7f300000000000000000000000000000000000000000001fffff7f300000000000000000000000008420003138b00010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011fb0001015a000000001b0018fe000002210175000007f000030000000000010002000000000000001900000019000000020000000200020000000000000000006c006c0000004400440000000000000001000000000000000000000000000001000c000000000000001ec9
2023-02-24 17:01:36.892 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-02-24 17:01:36.987 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA ADDRESS.
2023-02-24 17:01:36.988 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f7830220c3
2023-02-24 17:01:36.990 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.861 seconds (success: False)
2023-02-24 17:01:57.469 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.105.
2023-02-24 17:01:57.469 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-02-24 17:01:57.473 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-02-24 17:01:57.964 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70342000111f800000039003500300034003800450053004100470057003500300034001700170079000000003431302d30343032352d32333431302d30323033342d303058c7
2023-02-24 17:01:57.966 DEBUG (MainThread) [goodwe] Connected to inverter  G W 5 0 4, S/N: 9 5 0 4 8 E S A.
2023-02-24 17:01:57.969 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-02-24 17:01:58.730 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa17021811013805ba00030000002b05df00040000003c00000000000000000000000000000000000000220865005e138bfffff7f100000000000000000000000000000000000000000001fffff7f100000000000000000000000008440003138b00010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011f60001015a000000001b0018fe000002210176000007f600030000000000010002000000000000001900000019000000020000000200020000000000000000006d006d0000004400440000000000000001000000000000000000000000000001000c0000000000000097c2
2023-02-24 17:01:58.734 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-02-24 17:01:58.831 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA ADDRESS.
2023-02-24 17:01:58.831 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f7830220c3
2023-02-24 17:01:58.832 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.865 seconds (success: False)
2023-02-24 17:02:39.298 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.105.
2023-02-24 17:02:39.298 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-02-24 17:02:39.301 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
[... cut ...] 000000000000000000000000000001000c00000000000000b7ba
2023-02-24 17:08:06.938 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-02-24 17:08:07.040 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA ADDRESS.
2023-02-24 17:08:07.041 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f7830220c3
2023-02-24 17:08:07.043 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.866 seconds (success: False)

mletenay, 2023-02-24:
What is you inverter model ?
Since it seems it tries to talk ET protocol but I guess you have ES model ?

Diamantino-Op, 2023-02-24:
ESA5000

mletenay, 2023-02-24:
It seems to be a new model which is improperly identified by the code.
Isn't there one more request before the "Connecting to ET family inverter" line ?

Diamantino-Op, 2023-02-24:
no

mletenay, 2023-02-24:
Are you able to checkout and execute some python scripts ? (outside HA)

Diamantino-Op, 2023-02-24:
yes

mletenay, 2023-02-24:
Great. Can you reach me on our discord ? https://discord.gg/TaXyWXT

Diamantino-Op, 2023-02-24:
i'm in

mletenay, 2023-02-24:
I have released fixed underlying library, HACS integration is adjusted and will be released later.
In the meantime, you can re-download the HACS integration (enable "beta" version) and select "master".

closed 2023-02-24 by mletenay