mletenay/home-assistant-goodwe-inverter #167: Power Export Limit - ET series inverter

TomEkk33, 2023-02-26:
Hello @mletenay,

Power Export limit  value written by HA is always updated in related ET Inverter's register, which is OK.
[image]

However when Power Export limit is **disabled** in PV Master
[image]

HA does **not activate** "Power Export limit" during such write.

Please add option to explicitly enable/disable "Power Export limit" in HA. 
[image]

mletenay, 2023-03-09:
I'm somehow reluctant to add yet another switch to HA UI which vast majority of users do not need.
But as mentioned in https://github.com/home-assistant/core/issues/88963, you can turn the power limit switch by invoking "set_parameter" HA service with "grid_export" and 0/1

TomEkk33, 2023-03-10:
> get_parameter/set_parameter services

That's perfect. Thanks a lot.
Adding advanced features with call service as a general rule is excellent option. 
It does not introduce confusion to typical users.

closed 2023-03-12 by mletenay

petrrysavy74, 2023-04-25:
Exactly such a solution would be perfect for me. But I'm just a beginner. I don't know how to practically do the above procedure with get_parameter/set_parameter services "grid_export" setting on or off (0/1). Can you please help a beginner? Maybe send modified configuration files. I have the Goodwe experimental integration installed, it has really moved me forward in managing grid flows. The grid_export ON/OFF switch is important to me. I have a GW3600S-BP which is a retrofit inverter without a PV part and solves the balancing of the grid at the house output. Therefore, it has slightly different requirements than a regular FVE hybrid inverter. Thanks

TomEkk33, 2023-04-25:
@petrrysavy74

Below is a simple automation to turn off export power limit - be careful doing this.
```
alias: Goodwe_export_power_10kW
trigger: []
condition: []
action:
  - device_id: 47dcf029199ae1b3f5bcf99a51506445
    domain: number
    entity_id: number.grid_export_limit
    type: set_value
    value: 10000
  - service: goodwe.set_parameter
    data:
      device_id: 47dcf029199ae1b3f5bcf99a51506445
      parameter: grid_export
      value: 0
mode: single
```

Navigate in HA to "Settings -> Automations & Scenes -> create automation"
[image]
[image]
[image]

switch to "Edit in YAML" & paste the above code.

[image]

Then switch to visual mode. Update parameters in this mode if needed. Then save automation.

device_id - should be your inverter ID

To enable power limit to e.g. 100 W just change
```
    entity_id: number.grid_export_limit
    type: set_value
    value: 100

      parameter: grid_export
      value: 1
```

This automation was used for ET inverter type. I don't know other inverters.

Once gain setting incorrect values might give unpredictable results to your inverter. 
You can always check the results using PV Master/SolarGo application dedicated to your inverter

jl1990, 2023-05-27:
Not working for me neither. I have a GW5048-EM inverter, I added the experimental HACS version of this repo.
I can read values correctly, but setting them up does not seem to be working

mirecekd, 2023-05-29:
Works for me with v0.9.9.11
I have helper `input_boolean.fve_export_limit` which calls this automation:
```
alias: FVE nastavit grid_export
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.fve_export_limit
condition: []
action:
  - if:
      - condition: state
        entity_id: input_boolean.fve_export_limit
        state: "on"
    then:
      - service: goodwe.set_parameter
        data:
          device_id: putyoursdeviceid
          parameter: grid_export
          value: 1
    else:
      - service: goodwe.set_parameter
        data:
          device_id: putyoursdeviceid
          parameter: grid_export
          value: 0
mode: single
```
you can than put `input_boolean.fve_export_limit` to lovelaco or use (for example) in automation. I am using it to disallow export on negative spot prices -

disallow export
```
alias: FVE - vypnout přetoky - ENTRI výkup pod 0
description: ""
trigger:
  - platform: numeric_state
    entity_id: sensor.entri_aktualni_vykup
    below: -0.25
condition: []
action:
  - service: number.set_value
    data:
      value: "100"
    target:
      entity_id: number.grid_export_limit
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - service: input_boolean.turn_on
    data: {}
    target:
      entity_id: input_boolean.fve_export_limit
  - service: notify.telegram_mirecek
    data:
      message: >-
        Cena spotu od Entri je *menší* než -0.25 Kč/kWh - *vypínám* přetoky
      title: Zpráva od FVE
mode: single
```

allow export
```
alias: FVE - zapnout přetoky - ENTRI výkup nad 0
description: ""
trigger:
  - platform: numeric_state
    entity_id: sensor.entri_aktualni_vykup
    above: -0.25
condition: []
action:
  - service: number.set_value
    data:
      value: "10000"
    target:
      entity_id: number.grid_export_limit
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - service: input_boolean.turn_off
    data: {}
    target:
      entity_id: input_boolean.fve_export_limit
  - service: notify.telegram_mirecek
    data:
      message: >-
        Cena spotu je *větší* než -0.25 Kč/kWh - *zapínám* přetoky
      title: Zpráva od FVE
mode: single
```