mletenay/home-assistant-goodwe-inverter #168: Unable to add Inverter

pieterv1981, 2023-02-28:
Hi 

I have a GoodWe GW6K-DT with firmware 09.09 and ARM firmware 17 (According to goodwe support this is the latest firmware available)

So i want to add this in my HA and i'm using the ip it is getting from my local router (range 192.168.X.X) however I'm unable to connect to my inverter. Error Code: Failed to connect.  The HA and the inverter are in the same network.

What am I doing wrong?

mletenay, 2023-02-28:
Are you able to connect to the inverter with SolarGo mobile app ?

pieterv1981, 2023-03-04:
Hi Mletenay

Yes i can connect with the SolarGo app. And that's the strange thing. So i have no idea why i cannot connect with the HA 

Greetings

mletenay, 2023-03-05:
Increase the log level and check the output.
Put/adjust this fragment of configuration.yaml
```
logger:
  default: warning
  logs:
    components.goodwe: debug
    goodwe: debug
```

pieterv1981, 2023-03-09:
Hi Mletenay

Hereby the log:

2023-03-09 08:57:09.025 DEBUG (MainThread) [goodwe] Probing inverter at 192.168.230.100.
2023-03-09 08:57:09.027 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2023-03-09 08:57:10.028 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:10.028 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #1/10
2023-03-09 08:57:11.030 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:11.030 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #2/10
2023-03-09 08:57:12.032 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:12.033 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #3/10
2023-03-09 08:57:13.035 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:13.035 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #4/10
2023-03-09 08:57:14.041 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:14.041 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #5/10
2023-03-09 08:57:15.044 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:15.044 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #6/10
2023-03-09 08:57:16.047 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:16.047 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #7/10
2023-03-09 08:57:17.049 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:17.049 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #8/10
2023-03-09 08:57:18.051 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:18.051 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #9/10
2023-03-09 08:57:19.056 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-09 08:57:19.056 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #10/10
2023-03-09 08:57:20.058 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request aa55c07f0102000241 failed.
2023-03-09 08:57:20.059 DEBUG (MainThread) [goodwe] Probing ET inverter at 192.168.230.100.
2023-03-09 08:57:20.059 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-03-09 08:57:20.060 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-03-09 08:57:21.063 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:21.064 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #1/10
2023-03-09 08:57:22.065 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:22.065 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #2/10
2023-03-09 08:57:23.067 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:23.068 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #3/10
2023-03-09 08:57:24.071 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:24.072 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #4/10
2023-03-09 08:57:25.074 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:25.074 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #5/10
2023-03-09 08:57:26.075 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:26.076 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #6/10
2023-03-09 08:57:27.078 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:27.078 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #7/10
2023-03-09 08:57:28.081 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:28.081 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #8/10
2023-03-09 08:57:29.085 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:29.085 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #9/10
2023-03-09 08:57:30.087 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-09 08:57:30.087 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #10/10
2023-03-09 08:57:31.092 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request f70388b800213ac1 failed.
2023-03-09 08:57:31.093 DEBUG (MainThread) [goodwe] Probing DT inverter at 192.168.230.100.
2023-03-09 08:57:31.094 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
[... cut ...]
Pinging 192.168.230.100 with 32 bytes of data:
Reply from 192.168.230.100: bytes=32 time=3ms TTL=255
Reply from 192.168.230.100: bytes=32 time=2ms TTL=255
Reply from 192.168.230.100: bytes=32 time=2ms TTL=255
Reply from 192.168.230.100: bytes=32 time=2ms TTL=255

Ping statistics for 192.168.230.100:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 3ms, Average = 2ms

C:\>

So there is a reply and when i go to the ip address:

[image]

mletenay, 2023-03-09:
AFAIK, the IP 10.10.x.x is the IP which is used when you connect directly to the inverter wifi in AP mode (during wifi setup), so it's not real problem.
What is real problem is you inverter is not responding at all to any protocols (and it should respond to "DT" requests 7f03...).
Since you say SolarGo works (I assume you connect over wifi and not bluetooth) and have recent firmware, everything should work.
The only thing I can think of is some issue with your network setup ??
Is the HA machine on the same Wifi ?
Or if it is on the LAN segment, isn't there some firewall/router in between which does not pass UDP traffic ?

pieterv1981, 2023-03-10:
Hi 

When i connect with the SolarGO app it is over WIFI as BT is disabled. The HA is on the same lan segment as the inverter. The range is 192.168.230.X. There is a firewall but that firewall is filtering only traffic that is going outside. No internal traffic is filtered. I could make a rule that allows all data between the inverter an ha on the internal network but that will not do much as internal everything is allowed.

gr

pieterv1981, 2023-03-10:
[image]

100% sure :D

pieterv1981, 2023-03-15:
Hi

I checked the whole network and everything is working as it should. also again checked the static config of the inverter and it is as it should. Any idea what to do further?

wiljums, 2023-09-11:
@pieterv1981 have you solved the problem? Looks like I have the same issue with a GW15KN-DT + LAN-module, which was connected until yesterday

closed 2023-12-21 by mletenay

wiljums, 2024-05-08:
> @pieterv1981 have you ever found a solution for this?

After many tries mine connected.