mletenay/home-assistant-goodwe-inverter #169: SoC limit in ECO MODE

Bojkas1985, 2023-03-03:
Hi @mletenay,
from which FW version is it possible to set SoC limit in ECO MODE? I'm using 922, but the limit setting is not reflected there. Does this work in general on the whole ECO MODE, or is it necessary to use ECO CHARGE/DISCHARGE MODE? 
Do I understand correctly that if I set it to say 50% and switch to ECO MODE and the sun is shining, once the 50% value is reached all production will go to the grid for sale? Thank you

TomEkk33, 2023-03-03:
Hi @Bojkas1985,

I have ET inverter (Firmware: 04029-08-S11 / 02041-22-S00) and it works in ECO CHARGE/DISCHARGE MODE (**HACS only**).

Standard HA provides ECO MODE only.
The ECO MODE doesn't do this. It only switches to ECO mode but schedulers are not activated by HA.
See https://github.com/mletenay/home-assistant-goodwe-inverter/issues/151 for more details.

BTW,. You can always use standard Goodwe application to see results - PV Master for ET inverter.

mletenay, 2023-03-05:
On ET inverters, the Eco V2 (the one with SoC) is present since arm fw 19.
Your's 22 so you should have it.
As @TomEkk33 mentioned above - the standard Eco mode just changes the mode and does not adjust any of the eco mode schedules/powers/socs.
It's the Eco Charge mode which accepts the SoC and Eco Charge/Discharge with accepts power.

Bojkas1985, 2023-03-07:
Hey @mletenay , I was looking into PV Master. If I turn on Eco Charge Mode in HA, set to 60%, the inverter sets the charge but not the charge percentage, it stays there at 100%. Any advice on what I could be getting wrong? I have a 10kW ET.
[image]
[image]

TomEkk33, 2023-03-07:
Hi @Bojkas1985,
It works for me with the same ET inverter. I suppose you should first set %, then call the ECO charge mode.

Bojkas1985, 2023-03-07:
Ahoj @TomEkk33,
Yeah, that didn't help. I even tried resetting the SoC to 50% in PV master, but as soon as I sent the charge command, i.e. set the SoC to 60%, changed to charge mode, the SoC went back to 100%. Really strange.

mletenay, 2023-03-07:
Can you check the log files (set goodwe to debug) ?

Bojkas1985, 2023-03-08:
@mletenay Hi, i am not sure how get debug log. So, this is just copy all with debug after turn on goodwe debugging.
`2023-03-08 09:32:11.240 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.102.
2023-03-08 09:32:11.240 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-03-08 09:32:11.257 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-03-08 09:32:11.550 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703420002271000fe393031304b4554553231355730343733475731304b2d455420200009000900a3001600e930343032392d30392d53313130323034312d32322d5330307593
2023-03-08 09:32:11.657 DEBUG (MainThread) [goodwe] Connected to inverter GW10K-ET, S/N:9010KETU215W0473.
2023-03-08 09:32:11.708 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-03-08 09:32:12.219 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1703080a13270d6400220000048f0fba005100000cbd0000000000000000000000000000000000000202095d000813880000008c09a60004138a000000320972000313880000000e0001000000d1ffffffd60000000000000000095a0003138800010000000209a00002138a00010000000109760002138800010000000300000095000000350000002500000005000000f6000101af0000019640001f1d0f8b0bd7ff7afffff01400030000000100010000000000000001a0c60000003b0001a77a00002edb001c000016fa000000012186001c00006fc2003200006b690014000600050001000000000000000000000200098000020fd200008d12
2023-03-08 09:32:12.606 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-03-08 09:32:12.744 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff01010001019000120012000000430063000600000101000000000000000000000000000000000000000000000000f194
2023-03-08 09:32:12.939 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-03-08 09:32:14.988 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f7038ca0002dbbf3 in time (1s).
2023-03-08 09:32:14.988 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3 - retry #1/10
2023-03-08 09:32:15.466 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a00010064000a00000001ffe3fff6ffdfffb70399ffbbffd7ff8cffb713884aa2515c4a16f528ffffffe3fffffff6ffffffdfffffffb700000164000001290000010c00000399fffffe4bfffffebcfffffedefffffbe400010005a624
2023-03-08 09:32:15.488 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 3.830 seconds (success: True)
2023-03-08 09:32:15.662 DEBUG (MainThread) [goodwe.protocol] Sending: f703b090000337b0
2023-03-08 09:32:15.896 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703061703080a132b26bb
2023-03-08 09:32:16.749 DEBUG (MainThread) [goodwe.protocol] Sending: f703b996000155ec
2023-03-08 09:32:17.338 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70302006471ba
2023-03-08 09:32:17.753 DEBUG (MainThread) [goodwe.protocol] Sending: f703b798000136c7
2023-03-08 09:32:18.525 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-03-08 09:32:18.611 DEBUG (MainThread) [goodwe.protocol] Sending: f703b12c00017669
2023-03-08 09:32:18.718 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70302000af056
2023-03-08 09:32:18.887 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9ec00017435
2023-03-08 09:32:20.643 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f703b9ec00017435 in time (1s).
2023-03-08 09:32:20.643 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9ec00017435 - retry #1/10
2023-03-08 09:32:21.274 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-03-08 09:32:21.668 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-03-08 09:32:21.767 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007fff9c006400007943
2023-03-08 09:32:21.942 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-03-08 09:32:22.039 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007fff9c006400007943
2023-03-08 09:32:22.420 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-03-08 09:32:23.280 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1703080a13320d9f00220000049d0fc5005100000c7b00000000000000000000000000000000000002020944000a1387000000c4099b000613880000006b097100051388000000490001000001700000004b0000000000000000093c00031388000100000000099c00021388000100000002097400021388000100000002000000bc0000003f0000002f0000000500000120000101af0000019640001f060f7f0bd7ff7ffffff0a900030000000100010000000000000001a0c60000003b0001a77a00002edb001c000016fa000000012186001c00006fc2003200006b690014000600050001000000000000000000000200098000020f880000c36b
2023-03-08 09:32:23.443 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-03-08 09:32:23.517 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007fff9c006400007943
2023-03-08 09:32:23.601 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-03-08 09:32:23.819 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff01010001019000120012000000440063000600000101000000000000000000000000000000000000000000000000f455
2023-03-08 09:32:23.861 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-03-08 09:32:24.181 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a00010064000a00000001000e003a0022006c034e002400c10082006c13884aa2515c4a16f5280000000e0000003a000000220000006c0000016b000000fe000000e30000034e000001ba0000012500000100000003e0000100058a54
[... cut ...]
2023-03-08 09:35:22.581 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff010100010190001200120000004600630006000001010000000000000000000000000000000000000000000000002d95
2023-03-08 09:35:22.584 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-03-08 09:35:22.875 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a00010064000a00000001ff6bff86ff73fe630552ff05fef0fea7fee413894aa2515c4a16f578ffffff6bffffff86ffffff73fffffe630000021d000001b20000018200000552fffffda1fffffe31fffffe5cfffffa2c00010005586a
2023-03-08 09:35:22.877 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.691 seconds (success: True)`

mletenay, 2023-03-11:
@Bojkas1985 The log shows you selected plain Eco Mode. 
That one just sets the mode and takes the values stored already in inverter (e.g. via PvMaster).
It's only Eco Mode Charge / Discharge which also set the Soc/Power % values.

Bojkas1985, 2023-03-15:
Hi, I may have a problem with the mode setting. My options in automation stopped working. I can switch modes manually in the inverter operational mode option, but the automation says "Option Eco discharge mode not valid for select.inverter_operation_mode". Do you have any advice on how to build an automation to sell electricity to the grid? The way I have it at the moment is that I set the eco mode credentials to 1 so I don't discharge the battery unnecessarily, and then try to switch the mode (which doesn't work).

`alias: FVE - Automatizace prodeje spuštění
description: ""
trigger:
  - platform: time
    at: input_datetime.fve_prodej_zacatek
condition: []
action:
  - service: number.set_value
    data:
      value: "1"
    target:
      entity_id: number.eco_mode_power
  - service: select.select_option
    data:
      option: Eco discharge mode
    target:
      entity_id: select.inverter_operation_mode
  - wait_for_trigger:
      - platform: time
        at: input_datetime.fve_prodej_konec
  - service: select.select_option
    target:
      entity_id: select.inverter_operation_mode
    data:
      option: General mode
mode: single`

janchlebek, 2023-03-15:
Look here 

> Hi, I may have a problem with the mode setting. My options in automation stopped working. I can switch modes manually in the inverter operational mode option, but the automation says "Option Eco discharge mode not valid for select.inverter_operation_mode". Do you have any advice on how to build an automation to sell electricity to the grid? The way I have it at the moment is that I set the eco mode credentials to 1 so I don't discharge the battery unnecessarily, and then try to switch the mode (which doesn't work).
> * service: select.select_option
>   target:
>   entity_id: select.inverter_operation_mode
>   data:
>   option: General mode

Look here: https://github.com/mletenay/home-assistant-goodwe-inverter/releases/tag/v0.9.9.8
There was a breaking change. Modify your automations and change the option from "General Mode" -> "general", "Backup Mode" -> "backup" etc...).

Bojkas1985, 2023-03-15:
@janchlebek Thank you, my friend! I looked at the code and there is still eco_ in charge and discharge. 
You're selling on the spot? Do you have any interesting automation solutions?

TomEkk33, 2023-04-17:
Hello @mletenay,
It looks like v 0.9.9.11 does not honor 
number.eco_mode_power
value. 
The "ECO discharge mode" always discharges with 100% of available battery/inverter power.

What I mentioned here https://github.com/mletenay/home-assistant-goodwe-inverter/issues/169#issuecomment-1457688867
was related to Fast Battery Charge mode. Sorry for confusion.

TomEkk33, 2023-04-21:
Hello @mletenay, @Bojkas1985
I checked ECO Charge and Discharge modes once again today. All parameters settings work as expected. 
I don't know why 4 days ago didn't work for me. 

This issue can be closed as solved. Thanks.

closed 2023-12-21 by mletenay