mletenay/home-assistant-goodwe-inverter #170: GW4200-DNS-30 can't connect with HomeAssistant

Step-han, 2023-03-05:
Hello,

I've updateded my inverter from a 1 string GW1500-XS to a 2 string GW4200-DNS-30.

After installation al went fine, the inverter is online on the SEMS portal and app

In HA I can't connect on IP address, i've the right one. I've made it static. Only message I'll get is can't connect.

[image]

Is this inverter  just new and not yet available with this intergration? Tested die HA an HACS version.

Now I'm using this intergration https://github.com/TimSoethout/goodwe-sems-home-assistant but the data u get in HA is only per 5 minutes and its not local.

Hope to hear from you,

Step-han.

mletenay, 2023-03-05:
Can you connect to the inverter by SolarGo mobile app ?

Step-han, 2023-03-05:
> Can you connect to the inverter by SolarGo mobile app ?

I'll check it tomorrow, will let you know.

Step-han, 2023-03-06:
@mletenay 
I can connect with de SolarGo mobile app. See screenshots.

https://user-images.githubusercontent.com/54219681/223156072-1a2c9499-683d-4422-96d7-161017755d6a.png
https://user-images.githubusercontent.com/54219681/223156087-c352308a-e0f5-4d47-ae7b-6eaad5c722a8.png
https://user-images.githubusercontent.com/54219681/223156106-80979653-bd09-4159-8a79-4bf4d7f2e921.png
https://user-images.githubusercontent.com/54219681/223156121-0d2a05c5-1ab4-4b30-9cb9-f63e938adf2a.png

mletenay, 2023-03-07:
Setup/adjust the log levels in configuration.yaml like this:
```
logger:
  default: warning
  logs:
    components.goodwe: debug
    goodwe: debug
```
and attach the log output

Step-han, 2023-03-08:
@mletenay here to log output:

```
2023-03-08 11:24:07.180 DEBUG (MainThread) [goodwe] Probing inverter at 192.168.2.176.
2023-03-08 11:24:07.183 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2023-03-08 11:24:08.185 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:08.185 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #1/10
2023-03-08 11:24:09.187 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:09.187 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #2/10
2023-03-08 11:24:10.189 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:10.190 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #3/10
2023-03-08 11:24:11.193 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:11.194 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #4/10
2023-03-08 11:24:12.196 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:12.197 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #5/10
2023-03-08 11:24:13.199 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:13.199 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #6/10
2023-03-08 11:24:14.202 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:14.202 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #7/10
2023-03-08 11:24:15.204 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:15.205 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #8/10
2023-03-08 11:24:16.207 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:16.207 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #9/10
2023-03-08 11:24:17.211 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:17.211 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #10/10
2023-03-08 11:24:18.213 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request aa55c07f0102000241 failed.
2023-03-08 11:24:18.214 DEBUG (MainThread) [goodwe] Probing ET inverter at 192.168.2.176.
2023-03-08 11:24:18.215 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2023-03-08 11:24:18.217 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2023-03-08 11:24:19.218 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:19.218 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #1/10
2023-03-08 11:24:20.221 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:20.221 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #2/10
2023-03-08 11:24:21.223 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:21.223 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #3/10
2023-03-08 11:24:22.225 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:22.225 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #4/10
2023-03-08 11:24:23.227 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:23.227 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #5/10
2023-03-08 11:24:24.230 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:24.230 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #6/10
2023-03-08 11:24:25.232 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:25.232 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #7/10
2023-03-08 11:24:26.235 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:26.235 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #8/10
2023-03-08 11:24:27.237 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:27.238 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #9/10
2023-03-08 11:24:28.240 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70388b800213ac1 in time (1s).
2023-03-08 11:24:28.240 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1 - retry #10/10
2023-03-08 11:24:29.242 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request f70388b800213ac1 failed.
2023-03-08 11:24:29.244 DEBUG (MainThread) [goodwe] Probing DT inverter at 192.168.2.176.
2023-03-08 11:24:29.244 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
[... cut ...]
2023-03-08 11:24:49.300 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:49.300 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #9/10
2023-03-08 11:24:50.302 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to aa55c07f0102000241 in time (1s).
2023-03-08 11:24:50.302 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #10/10
2023-03-08 11:24:51.303 DEBUG (MainThread) [goodwe.protocol] Max number of retries (10) reached, request aa55c07f0102000241 failed.
```

[image]

mletenay, 2023-03-09:
Hmm, that is similar mystery than in the #168 with DT inverter.
You get no responses from the inverter at all.
Either Goodwe invented brand new 4th protocol (quite unlikely I guess), or there is something with your network ??
Is the HA machine on the same Wifi ? If not on Wifi but LAN, isn't there a firewall/router which does some filtering and does not properly propagate UDP traffic ?

Step-han, 2023-03-09:
> Hmm, that is similar mystery than in the #168 with DT inverter. You get no responses from the inverter at all. Either Goodwe invented brand new 4th protocol (quite unlikely I guess), or there is [...]

Nothing wrong with the network. TP-Link modem and deco m5 meshwifi accespoints (realty close to the inverter)

I tested with mesh wifi off and on filterd by the mac adress of the inverter.

The previous inverter GW1500-XS works like a charm.

mletenay, 2023-03-09:
UDP/8899 is used to communicate with inverter.
Can you sniff the traffic between the SolarGo and inverter to figure out why the app is working and HA not ?

Step-han, 2023-03-09:
I'll try tomorrow or Saturday.

Step-han, 2023-03-10:
Made a log:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/STEPHA~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/STEPHA~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

IPProto | SrcIP | SrcPort | DstIp | DstPort | UID | App | Proto | Status | Info | BytesSent | BytesRcvd | PktsSent | PktsRcvd
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
17 | 102151731 | 47100 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 67 | 1 | 1
17 | 102151731 | 42104 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 181 | 1 | 1
17 | 102151731 | 46363 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 77 | 1 | 1
17 | 102151731 | 55934 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 37 | 1 | 1
17 | 102151731 | 38412 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 115 | 1 | 1
17 | 102151731 | 33109 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 37 | 1 | 1
17 | 102151731 | 41706 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Active |   | 396 | 394 | 11 | 11
17 | 102151731 | 55209 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 115 | 1 | 1
17 | 102151731 | 40487 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 181 | 1 | 1
17 | 102151731 | 34880 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 37 | 1 | 1
17 | 102151731 | 57341 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 67 | 1 | 1
17 | 102151731 | 42769 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 37 | 1 | 1
17 | 102151731 | 40438 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 181 | 1 | 1
17 | 102151731 | 57522 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 77 | 1 | 1
17 | 102151731 | 37823 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 115 | 1 | 1
17 | 102151731 | 42773 | 1010100253 | 8899 | 10432 | SolarGo | UDP | Closed |   | 36 | 37 | 1 | 1
</body>
</html>

[image]

[image]

mletenay, 2023-03-10:
We need few of those payloads transmitted

Step-han, 2023-03-10:
Here some random payloads from the sniffer above.

[image]

Step-han, 2023-03-10:
> On a quick look it seems to be the proper protocol and HA should work. But I've noticed SolarGo is connected to 10.10. network, i.e. directly to the inverter and not to your 192.168. local network. So it again smells like the inverter is somehow not properly connected to net. Try to connect SolarGo to your local wifi and not the inverter directly.

Will do tomorrow!

Step-han, 2023-03-11:
Connect the inverter with home wifi via SolarGo app, still the same issue..

mletenay, 2023-03-12:
I'm afraid I have no other suggestion for you. 
The inverter seems to talk the proper protocol, so HA should work, but you receive absolutely no response to UDP packets.
Must be something specific to you network setup/firewall, you have to somehow figure it out.

Step-han, 2023-03-12:
There is no firewall that block UDP. 

Maybe the wifi-stick in the inverter is not good? I''ll get a other one to test.

mletenay, 2023-03-12:
Well, the HA talks exactly the same way as SolarGo.
Try using the app from the same network as HA and/or vice versa.
If previous inverter worked, try using the same IP if that has any impact.

timdemooij, 2023-05-04:
I have exactly the same problem with my GW4200-DNS-30. I have 5 Goodwe inverters and the GW4200-DNS-30 is the only one that doesn't work properly in HA. When I try to add it to HA, it kills my inverter connection and I have to do a complete reset of the inverter.
It looks like Goodwe changed something in their newest inverters (note that the the '-30' are relatively new Goodwe models)

I tried 2 different wifi-sticks, with the original firmware and updated firmware. But still the same issue.

My conclusion is that the GW4200-DNS-30 is not compatible with the Goodwe integration.

Step-han, 2023-05-04:
Finally someone who experiences the same. I even got a new wifi dongle under warranty, tested, exactly the same.

I think that this integration does indeed not work with the latest GoodWe inverter.

As a tip, I am now using this integration, which does work, via the sems-portal API: https://github.com/TimSoethout/goodwe-sems-home-assistant

Only there is 1 flaw, it refreshes the data once every 5 minutes.

mletenay, 2023-05-05:
The "official" integration will not be updated until someone figures out what is the problem.
The protocol seems to be unchanged, juts the inverter does not response, but it does response to SolarGo.
You have to figure out how/why ...

Mansjo, 2023-06-02:
My GW4200-DNS-30 won't connect either to Home Assistant. I can see the inverter in the SolarGo app, but it only gives "NA" values. Semsportal works fine. 

My GW3000-XS connects perfectly with my Home Assistant!

ewaldslms, 2023-10-16:
I have an GW3600-DNS-30 alongside a GW4K-DT
In my case, the DNS is also not connecting to Home assistant, the GW4K-DT is functioning properly.

Mansjo, 2023-10-20:
After I have opened a ticket on the GoodWe support site, I received an ARM Firmware update. Now I can connect to the inverter with Home Assistant!

https://support.goodwe.com/

ewaldslms, 2023-12-09:
> I have an GW3600-DNS-30 alongside a GW4K-DT I have exactly the same case, the GW3600 is not connecting to Home assistant, the GW4K-DT is functioning properly.

After an update by Goodwe i'm able to connect to GW3600-DNS30, but only once. The next day i have to restart/reboot Home Assistant and then HA is able to communicate with the convertor.

closed 2023-12-21 by mletenay

bartv42, 2024-06-10:
Did you guys figure this out? My 4200D-NS worked fine with Home Assistant until this morning, and now I can't get the integration to work again. I also noticed that the 'SolarWifi' network seems very spotty, and often disconnects - even when I'm close to the inverter. Could the wifi module be broken?

Piskatje13, 2025-05-07:
I have managed to add my GoodWe 4200 to HAS but next day it is "Unavailable"
[image]

Here are my settings:

[image]

After reloading it works again

[image]

But the next morning is the same story: the inverter is "Unavailable".  Every day is the same problem. Why?

Has anyone managed to get it working?

Piskatje13, 2025-05-07:
Tried to change to TCP 

[image]
 But it did not work

[image]

tomaswallentinus, 2025-05-07:
Most inverters shutdown when there is no sun to save energy. Is it that the issue you see?

Piskatje13, 2025-05-07:
> Most inverters shutdown when there is no sun to save energy. Is it that the issue you see?

I know that but the next morning the HAS should reconnect to the inventor, but it does not happen.