mletenay/home-assistant-goodwe-inverter #172: Support ET Serie 29,9kW with 2 Batteries

bastelfrosch, 2023-03-13:
Hello, is it possible to include the new GoodWe WR ET 29.9. Here the 2nd battery connection would have to be added. Furthermore, the inverter has 6 PV connections with 3 MPPT.
Thank you in advance

mletenay, 2023-03-13:
Oooo, that looks like some larger beast, no longer small residential inverter.
In theory yes, it might be possible to extend support for that, but there are few things to verify/test and I would need your cooperation there.
Open a related ticket in the underlying library - https://github.com/marcelblijleven/goodwe and it would be helpful if you'll setup python environment so you can execute some sample commands from there - the inverter_check.py

bastelfrosch, 2023-03-13:
Many thanks for the quick response.
I can't really program, only copy & paste works... :-) I'll get an environment installed and I'll probably be able to run scripts, but then that's it....

mletenay, 2023-03-13:
That should be good enough, just install python and checkout the library repo.
I'll give you instructions how to proceed, in best case we'll need just few tweaks.

bastelfrosch, 2023-03-14:
I have some errors in the inverter software.
E.g. the 2nd battery is not displayed correctly. I'm waiting for a new firmware to be installed. As soon as this is done, I'll let you know
Thank you in advance

mletenay, 2023-12-21:
ET >15 incl. 2nd battery is now properly supported.

closed 2023-12-21 by mletenay