mletenay/home-assistant-goodwe-inverter #174: kills my inverter connection

oziee, 2023-03-18:
I have used this integration a lot. and it seems that newer version now cause the inverter connection to die and the sensors become unavailable.

I will take some time to test which version last worked fine for me, but my GoodWe SBP inverter can no longer use this integration as is.

I have the Wifi kit and the LAN kit.. (got the LAN kit recently as I thought the problem was wifi module - but turns out to be this integration)

the problem is that the inverter stops sending data to HA and to SEMS.. but I can still log into the inverter admin page and change those settings.

The only fix is to completely shut down the entire inverter and restart.
After this SEMS and HA work again..but not for long.. soon enough they both stop working again.

I have disabled this integration for 2 weeks, and the inverter and SEMS data have been working fine.. and last night re-enabled this integration and soon enough all the data has stopped working again.

This integration must be transmitting a code to the inverter that the inverter is not liking and is causing the problem.. When I get a chance I will start to send the codes on by one from the newer integration code to see what changed from when this integration once worked fine, somewhere back around v0.9.9.3 ish

I see other issues reported here that are seeing similar results in the integration, so would be good to hear from some other users if they have seen and tried what I have done in the last month

samtrois, 2023-03-18:
Having the same issue on my 5048ES. 
Came here to make sure it wasn't just me. I'm new to all this so also thought it was my WiFi, and my HA.

Removed and re-added the integration doesn't help. 
Everything looks fine in PV Master and SEMS portal

tinuva, 2023-03-19:
@oziee I had a similar issue, well maybe still have it.

Initially I did not have any issues with the integration. Then some time later it started. I suspect, after I had the arm firmware updated to a newer version.

My connection would completely drop, so bad, that even pv master wouldn't connect and SEMS data update stopped.
Then to fix, I would completely shutdown the inverter (whole story) and do a wifi module reset (this would reset the AP password to 12345678). Only for it to work a few hours and then stop working again.

I eventually got tired of changing the AP password on the wifi device away from the default 12345678 to what I use each time I completely shutdown inverter and reset wifi module.
Then it kept on working.

So in my case the conditions are:
Device: wifi
ARM firmware: latest or very new to latest in 2022 Q3
Module AP wifi password: changed from default

I have heard rumours the wifi module is more stable than the lan module at best of times, who knows why but I don't have personal experience with that as I dont have the lan module.

oziee, 2023-03-19:
@tinuva correct I too have read that the wifi is more stable than the LAN..
I too have change the AP password.. lots of messing around :)

not sure about the firmware.. 
in the PV app It says I have inverter firmware 101014 
in the webpage is says Firmware version V1.1.2.1 which is probably the wifi dongle version

and yes when it dies out.. no PV master or PV go app connection.. SEMS connection.. just the web interface for wifi.. and have to do the whole shut right down and restart everything again..

tinuva, 2023-03-19:
Completely different versions here.

Inverter: 5048D-ES
Firmware sems : Version(FW/BMS)1242417/0
Firmware ha: 2424H
Wifi version: 1.0.3.8

samtrois, 2023-03-19:
Inverter: 5048D-ES
Firmware SEMS : Version(FW/BMS)6232315/0
Firmware HA: 410-04025-23 (2222C)
Wifi: ??

Plumpop1, 2023-03-25:
Hello friends.
 I work on the LAN module and no more outage.

samtrois, 2023-03-26:
I don't know if there was an update since, but it hasn't happened to me _(touch wood)_ since I posted here last week sometime.?

d-uhligde, 2023-04-15:
I have two inverters, an EM and an SDT G2 (both with WiFi).
The EM simply uploads its data to SEMS at any time, even with a poll every second.
The SDT G2 has a problem when I poll it too often via the Home Assistant. An interval of 300 seconds is not a problem. - Below this, no more data will be uploaded to the SEMS portal.

My guess: According to the documentation https://fccid.io/2ACSV-HF.../Users-Manual-3552381.pdf (chapter 2.6) there is a SOCKET B mode that supports the connection to multiple end devices. I suspect that the HF-LPT230 is active on the EM converter. But not with the SDT G2.
Unfortunately I could not send AT commands over the network and check the mode. Unfortunately, the HF-LPT230 does not have the associated TCP port open in order to establish a connection with the TCPUDPDbg tool (http://www.hi-flying.com/...ns-1/debug-tool-tcpudpdbg).

I currently have the original firmware installed on Hi-Fly (not from Goodwe) (http://www.hi-flying.com/download-center-1/firmware-1/download-item-hf-lpx30-firmware). - Unfortunately, I have not been able to send AT commands to the WiFi chip with this one either. Disadvantage of the Hi-Fly firmware: The PVMaster or SolarGo app no longer works.

Does anyone happen to have the original Goodwe firmware?

mpleite, 2023-08-06:
Now, for the first time (have the inverter dor 2 months now), the problem started here .
If i disable the integration, networks stays stable. As soon as i enable it ( or reload or retry after the permanent error), kills the wifi for some seconds, and on and on. Fails to ping, no report to sems portal, etc)
if i disable the integration, stays stable 
Info:
Home Assistant 2023.8.1 (via HAOS, full updated as today)
Supervisor 2023.07.1
Operating System 10.4

JochenHa, 2023-08-12:
Same issue here - inverter inresponsive, admin page for Wi-Fi  dongle is working. Inverter connected via lan.
I use home assistant most actual version and evcc add-on most actual version.
I will restart the inverter. All was working fine the last weeks...

oziee, 2023-08-12:
@JochenHa its been well talked about that the lan dongle causes lots of problem. I have both lan and wifi and the lan works 1 in very 100 or so ha calls and very problematic. Best to use the wifi dongle

JochenHa, 2023-08-13:
@oziee I'm afraid, Wi-Fi is not an option. HA connections works with LAN only.

Maybe the renewal of dhcp leases are not handled correctly? Seems to be the only "signal" using a very long timeframe.
I will try a fixed IP.

janchlebek, 2023-08-13:
> @oziee I'm afraid, Wi-Fi is not an option. HA connections works with LAN only.
> 
> Maybe the renewal of dhcp leases are not handled correctly? Seems to be the only "signal" using a very long timeframe.
> I will try a fixed IP.

I am running integration to HA on Wi-Fi and it is rock stable - not true it needs LAN. Check all DHCP settings and make reservation.

JochenHa, 2023-08-13:
@janchlebek interesting - I was not able to connect via Wi-Fi. Found some notes about it - maybe it's depending on the inverter type?

janchlebek, 2023-08-13:
> @janchlebek interesting - I was not able to connect via Wi-Fi. Found some notes about it - maybe it's depending on the inverter type?

Maybe - based on my experience and what I read the wifi module is considered to be more stable (at least on 10K-ET).

antonclaeys, 2024-01-25:
I have a 10K-ET as well. I'm running HA in docker (on Windows) on a server in the same subnet as the inverter (has fixed IP, set on the dongle itself). Using both the LAN as well as the WiFi module, my connection is very spotty. it's not possible to send commands to the inverter (for example to put it in ECO CHARGE MODE) because it fails in 95% of times.
I really don't know how to improve this.

craigcurtin-dev, 2024-02-13:
Yes the WIFI modules and what they support are region dependant - in Australia i am not aware of any of the wifi modules that support Modbus to the older SBP5000/3600 or the ES 5048 series inverters.

They do support the older AA55 protocol which is how SolarGo/SEMs/PV Master communicate with them

fortiko, 2024-06-26:
Hello! This behaviour just started on my 5048D-ES yesterday around 8:50 in the morning as well, with no changes to the setup and no updates either Home Assistant or the integration (non-HACS).

Symptoms:

1. Network connection is stable, WLAN dongle replies to pings every second
2. Add official HA GoodWe integration
3. Immediately timeouts start to occur (mostly in blocks of 3 seconds, which is the time the dongle needs to crash, reboot and re-connect to the WLAN)
4. The second I stop the integration and its initialisation process, pings become stable again and replies come in every second.

Is there really nothing that can be done about this? Perhaps the timing of the integration initialisation is too aggressive for newer firmware versions, and Goodwe added some kind of rate limiting?

I also wonder how I could detect whether Goodwe has updated the firmware on my inverter remotely around the times that the problems occurred?

I also tried to remove the official integration and then added the HACS branch, but the same problems occur.

Since I don't really care about the SEMS data, I also tried to block the inverter off the internet, to free up potential  remote connections to SEMS and perhaps make a local connection possible. No luck there either.

Thank you for your help!

```
64 bytes from 192.168.1.137: icmp_seq=2321 ttl=255 time=6.304 ms
64 bytes from 192.168.1.137: icmp_seq=2322 ttl=255 time=5.994 ms
64 bytes from 192.168.1.137: icmp_seq=2323 ttl=255 time=4.297 ms
64 bytes from 192.168.1.137: icmp_seq=2324 ttl=255 time=7.279 ms
64 bytes from 192.168.1.137: icmp_seq=2325 ttl=255 time=6.348 ms
64 bytes from 192.168.1.137: icmp_seq=2326 ttl=255 time=12.390 ms
64 bytes from 192.168.1.137: icmp_seq=2327 ttl=255 time=4.896 ms
64 bytes from 192.168.1.137: icmp_seq=2328 ttl=255 time=8.517 ms
64 bytes from 192.168.1.137: icmp_seq=2329 ttl=255 time=19.746 ms
64 bytes from 192.168.1.137: icmp_seq=2330 ttl=255 time=66.923 ms
64 bytes from 192.168.1.137: icmp_seq=2331 ttl=255 time=5.057 ms
64 bytes from 192.168.1.137: icmp_seq=2332 ttl=255 time=6.710 ms
64 bytes from 192.168.1.137: icmp_seq=2333 ttl=255 time=6.277 ms
64 bytes from 192.168.1.137: icmp_seq=2334 ttl=255 time=53.708 ms
64 bytes from 192.168.1.137: icmp_seq=2335 ttl=255 time=94.881 ms
64 bytes from 192.168.1.137: icmp_seq=2336 ttl=255 time=5.865 ms
64 bytes from 192.168.1.137: icmp_seq=2337 ttl=255 time=6.454 ms
64 bytes from 192.168.1.137: icmp_seq=2338 ttl=255 time=5.663 ms
64 bytes from 192.168.1.137: icmp_seq=2339 ttl=255 time=5.095 ms
64 bytes from 192.168.1.137: icmp_seq=2340 ttl=255 time=6.103 ms
64 bytes from 192.168.1.137: icmp_seq=2341 ttl=255 time=8.966 ms
64 bytes from 192.168.1.137: icmp_seq=2342 ttl=255 time=6.407 ms
64 bytes from 192.168.1.137: icmp_seq=2343 ttl=255 time=5.753 ms
64 bytes from 192.168.1.137: icmp_seq=2344 ttl=255 time=6.080 ms
64 bytes from 192.168.1.137: icmp_seq=2345 ttl=255 time=27.354 ms
64 bytes from 192.168.1.137: icmp_seq=2346 ttl=255 time=75.131 ms
64 bytes from 192.168.1.137: icmp_seq=2347 ttl=255 time=9.400 ms
64 bytes from 192.168.1.137: icmp_seq=2348 ttl=255 time=4.347 ms
Request timeout for icmp_seq 2349
Request timeout for icmp_seq 2350
Request timeout for icmp_seq 2351
64 bytes from 192.168.1.137: icmp_seq=2352 ttl=255 time=59.935 ms
64 bytes from 192.168.1.137: icmp_seq=2353 ttl=255 time=28.213 ms
64 bytes from 192.168.1.137: icmp_seq=2354 ttl=255 time=5.974 ms
64 bytes from 192.168.1.137: icmp_seq=2355 ttl=255 time=6.981 ms
64 bytes from 192.168.1.137: icmp_seq=2356 ttl=255 time=6.043 ms
64 bytes from 192.168.1.137: icmp_seq=2357 ttl=255 time=7.026 ms
64 bytes from 192.168.1.137: icmp_seq=2358 ttl=255 time=6.773 ms
Request timeout for icmp_seq 2359
Request timeout for icmp_seq 2360
Request timeout for icmp_seq 2361
64 bytes from 192.168.1.137: icmp_seq=2362 ttl=255 time=63.709 ms
64 bytes from 192.168.1.137: icmp_seq=2363 ttl=255 time=19.677 ms
64 bytes from 192.168.1.137: icmp_seq=2364 ttl=255 time=9.630 ms
64 bytes from 192.168.1.137: icmp_seq=2365 ttl=255 time=8.508 ms
64 bytes from 192.168.1.137: icmp_seq=2366 ttl=255 time=4.985 ms
64 bytes from 192.168.1.137: icmp_seq=2367 ttl=255 time=15.694 ms
64 bytes from 192.168.1.137: icmp_seq=2368 ttl=255 time=5.262 ms
Request timeout for icmp_seq 2369
Request timeout for icmp_seq 2370
[... 59 lines cut ...]
64 bytes from 192.168.1.137: icmp_seq=2430 ttl=255 time=9.158 ms
64 bytes from 192.168.1.137: icmp_seq=2431 ttl=255 time=10.193 ms
64 bytes from 192.168.1.137: icmp_seq=2432 ttl=255 time=9.052 ms
64 bytes from 192.168.1.137: icmp_seq=2433 ttl=255 time=10.174 ms
64 bytes from 192.168.1.137: icmp_seq=2434 ttl=255 time=8.842 ms
64 bytes from 192.168.1.137: icmp_seq=2435 ttl=255 time=4.777 ms
64 bytes from 192.168.1.137: icmp_seq=2436 ttl=255 time=9.882 ms
64 bytes from 192.168.1.137: icmp_seq=2437 ttl=255 time=4.739 ms
64 bytes from 192.168.1.137: icmp_seq=2438 ttl=255 time=10.069 ms
64 bytes from 192.168.1.137: icmp_seq=2439 ttl=255 time=6.652 ms
64 bytes from 192.168.1.137: icmp_seq=2440 ttl=255 time=9.768 ms
64 bytes from 192.168.1.137: icmp_seq=2441 ttl=255 time=14.274 ms
64 bytes from 192.168.1.137: icmp_seq=2442 ttl=255 time=9.689 ms
64 bytes from 192.168.1.137: icmp_seq=2443 ttl=255 time=13.058 ms
64 bytes from 192.168.1.137: icmp_seq=2444 ttl=255 time=8.371 ms
64 bytes from 192.168.1.137: icmp_seq=2445 ttl=255 time=4.109 ms
64 bytes from 192.168.1.137: icmp_seq=2446 ttl=255 time=9.495 ms
64 bytes from 192.168.1.137: icmp_seq=2447 ttl=255 time=5.741 ms
64 bytes from 192.168.1.137: icmp_seq=2448 ttl=255 time=6.378 ms
64 bytes from 192.168.1.137: icmp_seq=2449 ttl=255 time=5.082 ms
64 bytes from 192.168.1.137: icmp_seq=2450 ttl=255 time=10.832 ms
64 bytes from 192.168.1.137: icmp_seq=2451 ttl=255 time=6.136 ms
64 bytes from 192.168.1.137: icmp_seq=2452 ttl=255 time=4.641 ms
64 bytes from 192.168.1.137: icmp_seq=2453 ttl=255 time=9.839 ms
64 bytes from 192.168.1.137: icmp_seq=2454 ttl=255 time=9.006 ms

nzkiwi68, 2024-08-03:
I had terrible trouble with the inverter no longer accepting commands and also stopping updating SEMS portal after 1-2 days. The only way to recover was a full shutdown, solar, AC mains and batteries off - a complete cold restart. A Wi-Fi module reboot did not help. I'm using the Wi-Fi dongle.

I managed to get GoodWe to update the firmware to:

- GW5048D-ES
- Inverter firmware: 2323E
- Wi-Fi dongle
- Wi-Fi firmware: v1.0.3.8
- Wi-Fi Web: v1.0.2 

It's now rock solid stable. SolarGo still works, SEMS portal and HA just work. It has been going great now for many months without a reboot.

I use the HACS "GoodWe Inverter (experimental)" to switch from "General mode" supporting the house using battery power and I switch to charging the battery from grid mains at specific times using "Eco Charge Mode". I also do some great maths and calculate how hard to charge the batteries when on charge by manipulating the "Eco power power" percentage so that in a 3 hour charge period from mains the house batteries will come back to 100% with the lowest charge rate required.

srudin, 2024-09-26:
I was having exactly the same problem as @fortiko. For almost a year everything was working nicely, then about 2 months ago I started seeing the interruptions documented above.
Goodwe applied a firmware update to the inverter which did not help.
Then they gave me a firmware update for the wifi module (from 1.1.2.1 to 1.1.3.14) and after uploading this myself the interruptions were gone.

mpleite, 2024-09-26:
Do you have de FW?

srudin, 2024-09-27:
@mpleite You got it...

antonclaeys, 2024-09-29:
can we please share this firmware?

srudin, 2024-09-30:
Due to the large amount of requests I am getting: https://github.com/srudin/GoodweWifiFirmware

voodoopt, 2025-12-30:
> I have 2 inverters, a GW-5000-DNS-30, and a GW-8500-MS-30, both with identical wifi modules, both running exact same wifi firmware V1.1.3.14, the larger one is fine, the smaller inverter has the issues listed here, everything goes unavailable, it recovers after hours/days by itself.
> 
> So maybe it's more than just the wifi module firmware.

[image]

my module "wi-fi box"  Firmware version **V1.0.3.8** - Support says there are no upgrades.

[image]

@nojeffrey 
What is your dongle model and firmware?