mletenay/home-assistant-goodwe-inverter #175: GOODWE 3648d-es   entity

nokia321, 2023-03-22:
Hello. 
I'm from the Czech Republic, I don't speak the language, so I translate in a translator. I need advice on how to configure the GOODWE 3648D-ES photovoltaic inverter. The thing is that it does not show correctly the daily power production and furthermore it is not possible to insert the battery system into the tracking, because the entities are not included.  I don't really understand it, I don't know if I could set it up myself. How complicated is it to get this done.  Thank you Zdenek.

nokia321, 2023-03-22:
[image]

Plumpop1, 2023-03-25:
Zdeňku tvůj inverter neumí počítat baterii vstup a výstup. Jediná možnost je dopsat si entitu a vyčítat ze sems portálu.

Plumpop1, 2023-03-25:
Zdeňku musíš vyčítat tuhle hodnotu z atributů střídače. Dodělej si do HA Sems portal a tam ty hodnoty najdeš.
Střídač co máš tyhle statistiky nepředává do HA ale pouze na Sems portal.
Jinak jsme rádi že jsi z české republiky :-)

closed 2023-03-26 by nokia321