mletenay/home-assistant-goodwe-inverter #177: 5048-ES and Eco mode - what is going wrong with my set up?

PilgrimToHyperion, 2023-03-26:
Hello,

I've read through https://github.com/mletenay/home-assistant-goodwe-inverter/issues/164 but my issue is different.

I recently started using this experimental version (master) of the integration to make use of the eco_charge mode. Mostly it works wonderfully for me. I can change to eco_charge mode and set ECO SOC and ECO power mode without issue and it works at the correct values for the settings I set. 

The issue I'm facing is that every now and then the battery will not carry load after changing between eco_charge and general mode. E.g. DOD (On-Grid) is set to 65%, SOC is at 70%, mode is General. The battery should discharge to carry load until DOD is reached (so 35% of battery can be used). It does not and load is instead carried by the grid. Changing to other modes and back again (backup, eco, eco_charge, eco_discharge), does not help and load is still carried by the grid. I've also tried to see if switching off the battery and switching it back on works. It does not. Forcing off-grid mode (disabling grid physically) lets battery carry load, but when grid is available again, load is carried by battery. The only way to get it to behave as expected is to completely switch off the inverter and then when it is back on again, it works as expected and load is carried by the battery until DOD is reached. A day or two later, the behavior changes again and the battery will not carry load.

Please note that I've only seen this behavior since starting to use eco_charge mode. Previously (I was on HA official version) I was not changing settings via HA so I'm not sure whether it is a problem with my inverter, this integration etc.

I am on firmware 2424H, which Goodwe tells me is the latest. I'm not sure of the ARM version but I've logged a ticket to request that is updated as well. Where can I check the version?

Has anyone else experienced anything like this?

I've enabled the debug logging but I don't know where to find the output. Sorry, a bit of a noob. Happy to share screenshots of config, code, debug logs etc. if needed.

PilgrimToHyperion, 2023-04-03:
Okay, so after some further testing I can conclude the following:

* Using Solar Go and changing between eco modes and general mode works as expected with the battery taking load if required after changing back to general mode.
* If HA was used to set eco (charge) mode and is used to change it back to general mode, the issue this ticket was logged for occurs.
* If eco mode is set using Solar Go and HA is used to change it back to general mode, the issue this ticket is logged for does NOT occur.

It seems to be that something in the way that this integration sets eco_charge mode, prevents the inverter from fully returning to general mode. Perhaps some value is changed that is not returned to it's previous value when the mode is changed back to general?

Would really like to hear your thoughts on this @mletenay . I'm available on Discord as well if you want me to test or share more info.

tinuva, 2023-04-03:
have you tried to set the DoD again after changing back to General mode?

PilgrimToHyperion, 2023-04-03:
I do. I set it after changing back to general mode:

[image]

tinuva, 2023-04-03:
I am curious, does the Eco charge SoC value change when using Solargo?
Eg. before changing from eco charge to general, try setting the eco charge SoC to similar value you have for DoD.
So like, if you have DoD set to 65%, set the SoC to 35%. Just in case its actually stuck in eco charge mode still.

PilgrimToHyperion, 2023-04-04:
What worked yesterday isn't working today. Now matter what I do in Solar Go, I cannot get the battery to start discharging:

[image]

rinners, 2023-11-17:
I had issues with version 17, where before on version 13 I could change between Backup and General in HA and it would function as expected. With 17 it had some issue, I can't remember the exact symptoms, but I think it was something similar to what you are experiencing, so I decided to have them roll back to v13.

dl6dbh, 2024-02-15:
@PilgrimToHyperion 

Hello
Have you solved the problem ? can you switch from eco_charge_mode back to general mode?

PilgrimToHyperion, 2024-02-15:
@dl6dbh , hello. It seems to be working normally now, but also, I've not paid much attention to it as I've not received feedback on my question. It seems intermittent.

closed 2024-02-20 by PilgrimToHyperion