mletenay/home-assistant-goodwe-inverter #178: GoodWe GW6000-ES-20 - Post Firmware Upgrade Issue

ryanck123, 2023-03-28:
Hi 

My installer had to upgrade my firmware, and after having the system work correctly for about a week, it stopped. I'm currently on inverter firmware version V5 and ARM version V2 according to GoodWe support and PV Master app. Below is a snapshot of the error in the log:

_

Logger: goodwe.es
Source: /usr/local/lib/python3.10/site-packages/goodwe/es.py:199
First occurred: 09:31:04 (6 occurrences)
Last logged: 10:37:58

Error decoding firmware version .
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/goodwe/es.py", line 193, in read_device_info
    self.dsp1_version = int(self.firmware[0:2])
ValueError: invalid literal for int() with base 10: '\x00\x02'

_

Subsequent messages are failed to connect after retries, and no valid responses have been received.

The stock integration doesn't work, and I also tried reconfiguring WiFi and in a different HA instance.

Regards

mletenay, 2023-03-28:
I'll look into it, but if you could increase log level to DEBUG and attach here the response strings, it would help and allow me to write unit tests.

ryanck123, 2023-03-28:
[home-assistant_goodwe_2023-03-28T09-26-24.118Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/11087718/home-assistant_goodwe_2023-03-28T09-26-24.118Z.log)
Thank you very much. 

Based on some back and forth with GoodWe, it seems the model is still new and likely has a couple of issues. For some reason, I still get PV Master to work. Rolling back firmware isnt an option as I had an issue with the unit tripping on the old firmware if the grid is lost. I'm also going to use this tool later today to test connectivity. https://powerforum.co.za/files/file/60-goodwe_es-live-udp-data-by-cef-ver/. 

Will update as soon as I have managed to find something.

mletenay, 2023-03-28:
Your inverter seems to be "2nd generation" of "ES" inverters and it is quite possible it no longer talks the old "ES" protocol, but the newer "ET" protocol.
Could you try to explicitly specify family "ET" when adding integration to HA ?
(And attach the logs again ...)

mletenay, 2023-03-29:
It's difficult to debug it from HA.
Are you able to checkout the library sources from https://github.com/marcelblijleven/goodwe and run python script tests/inverter_check.py ?

ryanck123, 2023-03-30:
[invertercheck-none.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/11109775/invertercheck-none.txt)
[invertercheck-es.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/11109776/invertercheck-es.txt)

Here are the two output files, 1 running the test set to "None" and 1 set to "ES". Again ES and EM give responses, however, the rest don't. I have asked Goodwe on my firmware to reload it, in case that is the issue, since the firmware version query gives these special characters.

I can only assume that they have made significant changes in the latest firmware, given it worked before and PVMaster still works.

mletenay, 2023-03-30:
So it seems to talk the "ET" protocol, just they started to encode firmware versions differently.
I have committed a fix and protocol auto-detection.
Please checkout/pull the latest code and try again ...

mletenay, 2023-03-31:
You must have tested the old, not updated code.
The decoding now look like this - https://github.com/marcelblijleven/goodwe/blob/6895225c1a7780ff960140bd2690ecf9a2f127bb/goodwe/et.py#L347 and not as in the exception.

mletenay, 2023-03-31:
Hmm, it's probaly using the released version of library installed in python's pip.
Try to run the script directly from tests directory, then the sys.path.insert(0, '../../goodwe') on line 7 should ensure you're executing the local sources.

ryanck123, 2023-03-31:
Ok think I managed to get this right, see attached logs for both.
[inverter_check_es.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/11125244/inverter_check_es.txt)

[inverter_check_et.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/11125242/inverter_check_et.txt)

mletenay, 2023-03-31:
That looks good. 
Now uncomment the lines 42-45 and see if you get the sensor data.

ryanck123, 2023-03-31:
I have checked and can see both the real data and the settings, to confirm a more static value and I'm getting good data.

mletenay, 2023-03-31:
Great.
I have uploaded the changes, so if you (re) install the beta/master version of the integration in HACS, it should be working in HA properly now.
When you confirm it, I'll release a new version v0.9.9.10.

ryanck123, 2023-03-31:
I can confirm it works, I noticed the update and implemented it now. Thank you very much for making this work again for me.

ryanck123, 2023-03-31:
Maybe a FYI, the device controls hasn't populated in yet, but recall it didn't appear the first time immediately either.

closed 2023-03-31 by mletenay

antonclaeys, 2025-06-26:
did you receive a wifi lan kit 2.0 with this inverter? how did you enable modbus TCP IP? or are you using RTU?