mletenay/home-assistant-goodwe-inverter #182: Error on "inverter_scan.py", line 44

kendo55, 2023-04-09:
runnig inverter_scan.py:

```
PS C:\projekte\Python>  c:; cd 'c:\projekte\Python'; & 'C:\Users\hs3\AppData\Local\Microsoft\WindowsApps\python3.11.exe' 'c:\Users\hs3\.vscode\extensions\ms-python.python-2023.6.0\pythonFiles\lib\python\debugpy\adapter/../..\debugpy\launcher' '64580' '--' 'c:\projekte\Python\Goodwe\inverter_scan.py'
2023-04-09 09:24:58,122 __init__(633) - DEBUG: Using proactor: IocpProactor
2023-04-09 09:24:58,138 search_inverters(119) - DEBUG: Searching inverters by broadcast to port 48899
2023-04-09 09:24:58,138 _send_request(66) - DEBUG: Sending: 574946494b49542d3231343032382d52454144
2023-04-09 09:24:58,138 error_received(61) - DEBUG: Received error: [WinError 10057] Eine Anforderung zum Senden oder Empfangen von Daten wurde verhindert, da der Socket nicht verbunden ist und (beim Senden über einen Datagrammsocket mit einem sendto-Aufruf) keine Adresse angegeben wurde
2023-04-09 09:24:58,138 error_received(61) - DEBUG: Received error: [WinError 10022] Ein ungültiges Argument wurde angegeben        
2023-04-09 09:24:58,138 default_exception_handler(1771) - ERROR: Exception in callback _ProactorDatagramTransport._loop_reading()
handle: <Handle _ProactorDatagramTransport._loop_reading()>
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\proactor_events.py", line 573, in _loop_reading
    self._read_fut = self._loop._proactor.recv(self._sock,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\windows_events.py", line 463, in recv
    ov.WSARecv(conn.fileno(), nbytes, flags)
OSError: [WinError 10022] Ein ungültiges Argument wurde angegeben

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\proactor_events.py", line 579, in _loop_reading
    self._protocol.error_received(exc)
  File "C:\Users\hs3\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\goodwe\protocol.py", line 62, in error_received
    self.response_future.set_exception(exc)
asyncio.exceptions.InvalidStateError: invalid state
Traceback (most recent call last):
  File "c:\projekte\Python\Goodwe\inverter_scan.py", line 44, in <module>
    result = asyncio.run(goodwe.search_inverters()).decode("utf-8").split(",")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\hs3\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\goodwe\__init__.py", line 129, in search_inverters
    await response_future
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\proactor_events.py", line 533, in _loop_writing
    self._write_fut = self._loop._proactor.send(self._sock,
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\windows_events.py", line 566, in send
    ov.WSASend(conn.fileno(), buf, flags)
OSError: [WinError 10057] Eine Anforderung zum Senden oder Empfangen von Daten wurde verhindert, da der Socket nicht verbunden ist und (beim Senden über einen Datagrammsocket mit einem sendto-Aufruf) keine Adresse angegeben wurde

```
even after updating these modules:

pip install goodwe -U

pip install asyncio -U

I am also testing the program "inverter_test.py"

```
PS C:\projekte\Python>  & 'C:\Users\hs3\AppData\Local\Microsoft\WindowsApps\python3.11.exe' 'c:\Users\hs3\.vscode\extensions\ms-python.python-2023.6.0\pythonFiles\lib\python\debugpy\adapter/../..\debugpy\launcher' '64783' '--' 'c:\projekte\Python\Goodwe\inverter_test.py' 
2023-04-09 09:34:32,393 __init__(633) - DEBUG: Using proactor: IocpProactor
2023-04-09 09:34:32,409 discover(69) - DEBUG: Probing inverter at 192.168.1.180.
2023-04-09 09:34:32,409 _send_request(66) - DEBUG: Sending: aa55c07f0102000241
2023-04-09 09:34:32,530 datagram_received(49) - DEBUG: Received: aa557fc001824c30383038434757384b2d45542020200000000000000000000000000000000039383030304554553232325730303635000000000000000000000000000000004230323034312d31392d5330430cd1
2023-04-09 09:34:32,530 discover(77) - DEBUG: Detected ET/EH/BT/BH/GEH inverter GW8K-ET, S/N:98000ETU222W0065.
2023-04-09 09:34:32,530 _ensure_lock(124) - DEBUG: Creating lock instance for current event loop.
2023-04-09 09:34:32,530 _send_request(66) - DEBUG: Sending: f70388b800213ac1
2023-04-09 09:34:32,771 datagram_received(49) - DEBUG: Received: aa55f7034200011f4000fe393830303045545532323257303036354757384b2d455420202000080008009f001300cf30343032392d30382d53313130323034312d31392d5330304ea8
Traceback (most recent call last):
  File "c:\projekte\Python\Goodwe\inverter_test.py", line 30, in <module>
    f"- Version: {inverter.software_version}"
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ET' object has no attribute 'software_version'
```
without line 30 

#f"- Version: {inverter.software_version}"

"inverter_test.py"  works

kendo55, 2023-04-14:
Thanks for your work and your answer. 

You mean, if I search the appropriate functions in the library I would by able to do it myself?

mletenay, 2023-04-14:
Sure, it's this function - https://github.com/marcelblijleven/goodwe/blob/29de1c4303fffd3e984eb3314db4b5085fbaf334/goodwe/__init__.py#L113
It's UDP broadcast to port 48899.

kendo55, 2023-04-14:
I tried this:

````
from __future__ import annotations

import binascii
import goodwe
import sys

import asyncio
import logging
from typing import Type

from .dt import DT
from .es import ES
from .et import ET
from .exceptions import InverterError, RequestFailedException
from .goodwe import GoodWeXSProcessor, AbstractDataProcessor, GoodWeInverter
from .inverter import Inverter, OperationMode, Sensor, SensorKind
from .model import DT_MODEL_TAGS, ES_MODEL_TAGS, ET_MODEL_TAGS
from .protocol import ProtocolCommand, UdpInverterProtocol, Aa55ProtocolCommand

logger = logging.getLogger(__name__)

# Inverter family names
ET_FAMILY = ["ET", "EH", "BT", "BH"]
ES_FAMILY = ["ES", "EM", "BP"]
DT_FAMILY = ["DT", "MS", "NS", "XS"]

# Initial discovery command
DISCOVERY_COMMAND = Aa55ProtocolCommand("010200", "0182")

# supported inverter protocols
_SUPPORTED_PROTOCOLS = [ET, DT, ES]

async def search_inverters() -> bytes:
    """Scan the network for inverters.
    Answer the inverter discovery response string (which includes it IP address)
    Raise InverterError if unable to contact any inverter
    """
    logger.debug("Searching inverters by broadcast to port 48899")
    loop = asyncio.get_running_loop()
    command = ProtocolCommand("WIFIKIT-214028-READ".encode("utf-8"), lambda r: True)
    response_future = loop.create_future()
    transport, _ = await loop.create_datagram_endpoint(
        lambda: UdpInverterProtocol(response_future, command, 1, 3),
        remote_addr=("255.255.255.255", 48899),
        allow_broadcast=True,
    )
    try:
        await response_future
        result = response_future.result()
        if result is not None:
            return result
        else:
            raise InverterError("No response received to broadcast request.")
    except asyncio.CancelledError:
        raise InverterError("No valid response received to broadcast request.") from None
    finally:
        transport.close()

search_inverters()

````
But the .xxx  like Import ".goodwe" could not be resolved

[image]