mletenay/home-assistant-goodwe-inverter #183: setting "eco discharge mode" or "eco charge mode"  does not work.

crispe, 2023-05-06:
On this inverter:
GW5048D-ES
by GoodWe
Firmware: 2121C / None

Home Assistant 2023.5.2
Supervisor 2023.04.1
Operating System 10.1
Frontend 20230503.3 - latest

It fails, here is the log:

Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/goodwe/select.py:131 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 18:50:04 (1 occurrences) 
Last logged: 18:50:04

[547355618192] Aa55ReadCommand.__init__() takes 3 positional arguments but 4 were given
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1849, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1889, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 981, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 106, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 131, in async_select_option
    await self._inverter.set_operation_mode(
  File "/usr/local/lib/python3.10/site-packages/goodwe/es.py", line 322, in set_operation_mode
    await self.write_setting('eco_mode_2_switch', 0)
  File "/usr/local/lib/python3.10/site-packages/goodwe/es.py", line 250, in write_setting
    register_data = await self._read_from_socket(Aa55ReadCommand(self.comm_addr, setting.offset, 1))
TypeError: Aa55ReadCommand.__init__() takes 3 positional arguments but 4 were given

Has anyone encountered this?

JoshC1994, 2023-07-05:
Yes im encountering the exact same issue, with the same inverter GW5048D-ES. Have you had any luck?

### Failed to call service select/service_option. Aa55ReadCommand._init_() takes 3 positional arguments but 4 were given ###

tinuva, 2023-07-05:
@crispe

You inverter firmware `2121C` is not the latest. You need to be on at least `2424H`

You need to open 2 tickets with Goodwe.
1. To upgrade inverter firmware
2. To upgrade arm firmware

Then try again.

For me, eco charge and eco discharge works 100% from Home Assistant using `2424H`

JoshC1994, 2023-07-06:
@tinuva  I reached out to Goodwe and they updated me to 2222C and told me that it is the newest version. 

I'm still having the same issue and can't get it to work, In the meantime I have raised another ticket for an update.

JoshC1994, 2023-07-09:
@tinuva No luck I reverted back to v0.9.9.7 (Furthest back I could go) and still not working. This sucks :(  

[image]
 version

JoshC1994, 2023-08-07:
@mletenay Are you able to shed some light?    Goodwe say that 2222C is the newest firmware and also that my ARM is fully updated. and i am still getting this error, I've tried downgrading an i still run across issues, best so far is version v0.9.9.9 but i cannot seem to get it to properly discharge. 

**Failed to call service select/service_option. Aa55ReadCommand.init() takes 3 positional arguments but 4 were given**

JoshC1994, 2023-08-14:
@tinuva Are you able to screenshot your inverter firmware version. Goodwe don't seem to be playing ball saying I have the newest version but they just updated me to 2222C, and heaps of people are running 2424H. so if i have some evidence I can maybe get a better response.

tinuva, 2023-08-14:
> @tinuva Are you able to screenshot your inverter firmware version. Goodwe don't seem to be playing ball saying I have the newest version but they just updated me to 2222C, and heaps of people are running 2424H. so if i have some evidence I can maybe get a better response.

[image]

wishie, 2023-08-20:
I can't even get GoodWe to respond to any ticket.. so how am I supposed to get the inverter updated?

JoshC1994, 2023-08-20:
So I was able to get my issue resolve and Yes!, all it took was an Software Update to 2424.  Now convincing Goodwe to admit that I wasn't up to date well that proved difficult. They initially updated me to Version 1818 (6181819), when I asked again, I was informed I was "100% up-to-date".  I reached out via other means when I was not getting a response and I was then updated to Version 2222C (6122219). I then persisted and told them that I wish to be updated to 2424, Now I went back and forth with them for nearly 3 weeks in regards to this. At one point I had my SEMS portal reporting 2424 but not the Inverter directly (I hope not) but me being so persistent/annoying I believe there was some foul play and they just changed what SEMS was reporting quieten me. With the help of @tinuva and a few others I was able to provide them with evidence that others were running 2424H, I ended up speaking on the phone with Mike Zhang _(Goodwe AU/NZ Solar Installers Facebook Group)_  and they accepted that YES I was infact running older firmware and I would be updated. 10min later I was running Version 2424J and I was able to get my inverter to manually Charge & Discharge using eco_mode and better yet this was on version v0.9.9.9 and now also confirm working on 0.9.9.11. 

Hopefully this helps anyone that is having these issues. 

* Next step, to block inbound connections from Goodwe on my Modem, to stop any foul play. Maybe I'm being cynical but Geopolitical tensions are high and there has been a lot of talk in Government here in Australia lately regarding the security of our grid, and the impact that foreign inverters may have if relied on too heavily. Funnily Goodwe was the first to put out a statement saying they have servers hosted locally in Australia and "operate independently" . Could be seen as admission of guilt or all round nice guy. ¯\_(ツ)_/¯       _(https://reneweconomy.com.au/we-operate-independently-china-inverter-giant-answers-coalition-pv-scare-campaign/)_

@tinuva - Thanks Heaps, this was the firing gun that helped me resolve this with Goodwe. 

@wishie - Be persistent try Ringing, reaching out via Facebook or their WhatsApp. Mine was a 3 week Ordeal.

wishie, 2023-08-20:
Interestingly enough, I emailed Mike Zhang last night, as I had a previous conversation with him. He has asked for my inverter serial number, so I assume he is ready to do the update

wishie, 2023-08-22:
I looked in PV Master and my GW5048-EM inverters are both showing 1010B as the firmware.. Mike Zhang said he was going to upgrade them the other day.. I will send off another email.

wishie, 2023-08-23:
I am being told that 1010B is the latest firmware for the -EM series inverters. Can anyone confirm this?

JoshC1994, 2023-08-23:
@wishie - I started a chat on [Goodwe solar and battery Inverter owners](https://www.facebook.com/groups/777057633698270/)

shiner66, 2023-09-16:
> @crispe
> 
> You inverter firmware `2121C` is not the latest. You need to be on at least `2424H`
> 
> You need to open 2 tickets with Goodwe.
> 
> 1. To upgrade inverter firmware
> 2. To upgrade arm firmware
> 
> Then try again.
> 
> For me, eco charge and eco discharge works 100% from Home Assistant using `2424H`

I am running 2424H but still no luck with the same error on my 5048-ESA

AGCBro007, 2024-01-23:
> I am being told that 1010B is the latest firmware for the -EM series inverters. Can anyone confirm this?

I am actually running version 1111H on my EM series inverter. Not sure about the ARM version, don't know how to check it. 
My HA integration works, but with the SolarGo app I am able to define up to 4 Battery Power Groups in the economic mode, each group with their own definition of begin & end time, repeat schedule, Rated Power, charge/discharge, Charge cut-off SOC. This I cannot find on the HA(CS) integration. Is this a known issue?

I applied for the facebook group, hope to find more info there.

wishie, 2024-01-23:
I'm stuck on 1010B with the local goodwe people claiming that's the latest.

mletenay, 2024-02-14:
The "takes 3 positional arguments but 4 were given" bug was fixed quite some time ago, closing the ticket.
If the Eco Mode does not work from any other reason, open new ticket please.

closed 2024-02-14 by mletenay

JoshC1994, 2024-02-14:
@mletenay I am still experiencing this issue with a EM inverter as of friday.  I now am unable to test as i requested Goodwe Australia to update my ARM & DSP and they have bricked my inverter and now wifi dongle is stuck in a boot loop and won't work 😑.