mletenay/home-assistant-goodwe-inverter #185: Nefunkční GoodWe Experimental

Bojkas1985, 2023-05-15:
Ahoj @mletenay  
mně a dvou dalším kamarádům přestala fungovat integrace. Mně to padlo dnes kolem 9h a pak přestalo odesílání dat do HA. Žádnou aktualizaci jsem nedělal, kamarád to zrovna dnes aktualizoval, všichni jsme momentálně na 2023.5.x. Zkoušel jsme integraci znovu stáhnout, nově načíst, ale píše mi to chyby, že se to nenačetlo, v logu pak jen viz screen.
[image]

mletenay, 2023-05-15:
Nezmenila sa nahodou IP adresa invertera ? PvMaster/SolarGo funguje ?

Bojkas1985, 2023-05-15:
@mletenay no, to je teoreticky možné. :D Kde se dá vyčíst IP adresa, kterou jsem na začátku zadával?

Bojkas1985, 2023-05-15:
@mletenay Tak nic, omlouvám se, byla to jen blbá shoda náhod, u mě to byla IP, u druhého kamaráda něco jiného, ale jede mu to, tak jsem myslel, že se to týká více lidí.. Uzavírám, klidně to smaž. Jinak, IP adresy se dají zjistit tady. :) /config/.storage/core.config_entries

closed 2023-05-15 by mletenay