mletenay/home-assistant-goodwe-inverter #186: Add SEC1000 support - It's impossible to set the grid export limit when a SEC1000 is connected

MrOzzOnline, 2023-05-16:
When you have multiple Goodwe hybrid inverters you can use a SEC1000 to balance out power between the grid and various inverters and batteries using a SEC1000 or a SEC1000s. 
This device will override the grid export limit of each inverter and has it's own power limit settings. At the moment it's impossible to control the grid export using home assitant if you have a SEC1000 installed, thus it's not possible to stop injecting when prices go negative.

There already is a Python module which access the SEC1000 but it doesn't include the Power Limit setting.
https://github.com/GTEC-UDC/goodwe-sec1000-info

vikicv, 2023-05-22:
I have same experience, recently installed SEC1000s with two ET inverters connected via RS485. SEC1000s is overriding all settings sent to inverters like it is inverter operation mode, grid export limit.... thus forced discharging batteries into the grid during highest energy spot price (eco discharge mode) does not have any effect.

MrOzzOnline, 2023-05-30:
This is costing me money. Yesterday I paid €5 for injecting current and I could not turn it off.
So ... I'm willing to pay someone to fix this. Requirements:
* Add SEC1000 support to the goodwe custom integration
* Make sure the power limit can be set from HA (same as the power limit in the Promate software) or make sure you can enable or disable electricity export for the SEC1000 using HA.
* Bonus: enable or disable charging the connected batteries using HA - not sure if this is possible.

TomEkk33, 2023-06-04:
Hello @MrOzzOnline,

I'll also have two inverters soon with SEC1000 ...
Can you try to uncheck "Export Enab" setting in SEC1000?
[image]

In SEC1000 instruction manual there is a sentence:
[image]

So "Export Enab" is optional and maybe "uncheck" will help to stay with original "Power Export" limit set in individual ET inverters
Let me know if this works.

vikicv, 2023-06-04:
> Hello @MrOzzOnline,
>
> I'll also have two inverters soon with SEC1000 ...
>
> Can you try to uncheck "Export Enab" setting in SEC1000? [image]
>
> In SEC1000 instruction manual there is a sentence: [image] [...]

Deat TomEk33, the "export enab" is enabling export power limit function... once you tick it, the SEC1000s will set via RS485 command the power export limitation in all connected inverters, if you have a two inverters, the limit will be set to the half in each (10kW limit in SEC1000s = 5kW limit set in both inverters). Once you untick "export enab" this will disable the power export limit function in all connected inverters. This can be seen / verified via SolarGo app accessing each individual inverter.

TomEkk33, 2023-06-04:
OK, thanks for this info. So SEC1000 kills all the control in ET inverter.
What about "Fast charge" option in ET inverter? 
Will it charge the battery connected to certain ET inverter when SEC1000 is in the system?

Bojkas1985, 2023-06-08:
Hey @mletenay ,
do you plan on adding SEC1000 support to the integration? Alternatively, is it possible to add two identical ET inverters to the HA and read data from them and control them?

mletenay, 2023-06-08:
No, I do not plan to add support to SEC1000. I do not own such device and have no means to try/test anything.
You are of course welcome to implement it yourself and submit a PR, but unless you or someone does that, there will be no new code.
Regarding multiple inverters, yes, I believe the integration does already support that, there are few users out the (with ES) who have multiple inverters.

MrOzzOnline, 2023-06-20:
> > Hello @MrOzzOnline,
> > I'll also have two inverters soon with SEC1000 ...
> > Can you try to uncheck "Export Enab" setting in SEC1000? [image]
> > In SEC1000 instruction manual there is a sentence: [image]
> > [image]
> > So "Export Enab" is optional and maybe "uncheck" will help to stay with original "Power Export" limit set in individual ET inverters Let me know if this works.
> 
> Dear TomEk33, the "export enab" is enabling export power limit function... once you tick it, the SEC1000s will set via RS485 command the power export limitation in all connected inverters, if you have a two inverters, the limit will be set to the half in each (10kW limit in SEC1000s = 5kW limit set in both inverters). Once you untick "export enab" this will disable the power export limit function in all connected inverters. This can be seen / verified via SolarGo app accessing each individual inverter.

I've tried a few configurations and there doesn't seem to be a way to bypass the SEC1000 and to adjust the export power.

At the moment I'm looking at the remote shutdown and DRED function of the inverter. 
The problem with the remote shutdown is that it cuts all production, not just injection of power.
About the DRED function I'm not sure if it's meant to control external devices or if it is used to control the power of the inverter.

Samorja, 2023-12-16:
> I've tried a few configurations and there doesn't seem to be a way to bypass the SEC1000 and to adjust the export power.
>
> At the moment I'm looking at the remote shutdown and DRED function of the [...]

Hello MrOzzOnline, do you have any news on this topic? I have the same issue, running a setup of 10x10kw ET inverters shutting down modbus connection, then i need to push or pull power from grid, but this is no solution to your problem.

closed 2023-12-21 by mletenay

MrOzzOnline, 2024-02-05:
> Hello MrOzzOnline, do you have any news on this topic? I have the same issue, running a setup of 10x10kw ET inverters shutting down modbus connection, then i need to push or pull power from grid, [...]

No, this remains unresolved. There are a few days in a year when prices are negative and I'm exporting. I'm not loosing a lot of money because of it. One option I considered is switching off grid using an automated switch, but it would take a few years to pay back the switch. I don't think it's worth the effort.
Goodwe now has extended their ET series upto 30kW. There will even be fewer cases where the Sec1000 will be needed. I don't have a lot of hope to get a good solution for this.