mletenay/home-assistant-goodwe-inverter #188: Eco mode charge/discharge on GW3648D-ES

petrkac, 2023-05-21:
Hi,
I have GW3648D-ES with Firmware: 1818A / None. 
When I choose Inverter operation mode either Eco discharged or Eco charger mode I just receive error message "Failled to call servrice select/select_option. Aa55ReadCommand__init__() takes 3 positional arguments but 4 were given" and mode does not change. It look that "eco mode power" does not take effect (in PV master is 10%). 
I may not use Eco mode setings, because my GW does not save the ECO parameters, after few hours it reseted do some default parameters (something like 10% Eco mode power).
When I use "backup mode" it charges only on 10% power (probably GW use the same setting like for eco mode).
Is it please posible to correct the behavior of eco charge/discharge mode?
Thank you
Petr

tinuva, 2023-05-21:
You need a newer Firmware and newer arm software on your inverter. On 2424H eco mode works 100%

Best is to open 2 tickets to goodwe. 1 to update Firmware to latest. 1 to update arm to latest.

Your other option is to look for a much older version of this integration that worked with that old Firmware you have

petrkac, 2023-05-23:
Thank you, upgraded to Firmware: 2525J / None and everything is OK now.

shiner66, 2023-09-16:
> You need a newer Firmware and newer arm software on your inverter. On 2424H eco mode works 100%
>
> Best is to open 2 tickets to goodwe. 1 to update Firmware to latest. 1 to update arm to latest.
>
> Your [...]

I am running 2424H but it isn’t working, what can I do?

mletenay, 2024-02-14:
The "takes 3 positional arguments but 4 were given" bug was fixed quite some time ago, closing ticket.

closed 2024-02-14 by mletenay