mletenay/home-assistant-goodwe-inverter #191: House Consumption is wrong on low load

smileyman-1, 2023-05-28:
If consumption is low and there is no pv-input (at night), the House Consumption value is incorrect.
[image]
At night, a generation of 15 watts is detected in this way. 

On-grid Export Power shows the same wrong value. (In PV-Master the value is correct.)
On-grid Power and Plant Power, on the other hand, usually - but not always - shows at this time the correct value. (But this values are always positiv and do not switch to minus for importing)
[image]

When the load is higher House Consumption shows the correct values.
[image]

Inverter: GW3068-EM

Paja-git, 2023-10-11:
The house consumption is a calculated value. PV1Power+PV2Power+BaterryPower-ActivePower. ModBus registers are read sequentially, the inverter's probes have some precision also. So, small calc errors can happen.

closed 2023-12-21 by mletenay

nuitari83it, 2025-02-11:
Same "problem" here.
What is exactly the difference between "house consumption" and "plant power"? What is exactly measuring the second one?

Paja-git, 2025-02-14:
> The latest released Goodwe integration for HA with a GW3048-EM inverter

OK, it explains a lot - sensor is listed in the es.py. The Plant power is calculated value:  the on-grid + backup power sensors. The House consumption calcutates PV1,PV2, Battery, Export.

nuitari83it, 2025-02-15:
thanks for the explanation, so basically it doesn't count the photovoltaic production influence