mletenay/home-assistant-goodwe-inverter #194: ET + eco charge power not accepted

bakerboy908, 2023-06-03:
Firmware: 04029-10-S11 / 02041-23-S00
Hello,

in echo chrage mode the power is not being accepted, when in echo charge mode the inverter always charges the battery at maximum charge current.
Logs
```
2023-06-03 10:31:27.199 DEBUG (MainThread) [custom_components.goodwe.select] Setting eco mode power to 0
2023-06-03 10:31:27.200 DEBUG (MainThread) [goodwe.protocol] Sending: f710b9bb00060c0000173bff7f000000640000771e
2023-06-03 10:31:27.258 DEBUG (MainThread) [goodwe.protocol] Received: aa55f710b9bb000601e4
2023-06-03 10:31:27.259 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9c3000145fc
2023-06-03 10:31:27.501 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-06-03 10:31:27.502 DEBUG (MainThread) [goodwe.protocol] Sending: f706b9c30000483c
2023-06-03 10:31:27.752 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b9c30000483c
2023-06-03 10:31:27.753 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9c9000165fe
2023-06-03 10:31:28.001 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-06-03 10:31:28.002 DEBUG (MainThread) [goodwe.protocol] Sending: f706b9c90000683e
2023-06-03 10:31:28.253 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b9c90000683e
2023-06-03 10:31:28.254 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9cf000185ff
2023-06-03 10:31:28.503 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-06-03 10:31:28.504 DEBUG (MainThread) [goodwe.protocol] Sending: f706b9cf0000883f
2023-06-03 10:31:28.753 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b9cf0000883f
2023-06-03 10:31:28.754 DEBUG (MainThread) [goodwe.protocol] Sending: f706b79800037b06
2023-06-03 10:31:29.007 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b79800037b06
2023-06-03 10:31:29.008 DEBUG (MainThread) [goodwe.protocol] Sending: f710b99700020400010000a8f9
2023-06-03 10:31:29.253 DEBUG (MainThread) [goodwe.protocol] Received: aa55f710b9970002c1ee
2023-06-03 10:31:30.443 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-03 10:31:30.748 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1706030a1f1d0dc40014000002c20ab40024000003da0000000000000000000000000000000000000202098d001d1382ffffff8f0957001a1387000000c209710021137ffffffdf30001fffffe69fffffca30000000000000000096e0005138200010000002e094b000d13880001000000b1095f000c137f0001000000bf0000001c0000000000000019000001d600000010000701a8000001ce40201ee20f720863ffa5fffff8580003000000090001000000000000000010b50000003500000f490000021c001f000000df000400000fcb002c000005a6001f000004c3000b0004000000010000000000000000000002000800000204db00001fe1
2023-06-03 10:31:30.749 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-06-03 10:31:30.839 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff0192000100cc001900190000002f006300040000012200000000000000010111031210120e1100cf00cb0d2e0d240ef4
2023-06-03 10:31:30.840 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-03 10:31:31.128 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a0001005e000a01110001ffd7009bfde1fe52fc31fe93002efcabfe0313874803db004820c580ffffffd70000009bfffffde1fffffe52fffffe46fffffed9ffffff14fffffc31fffffe25000001b9fffffd57fffff9c200010005f488
2023-06-03 10:31:31.128 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.686 seconds (success: True)
2023-06-03 10:31:36.443 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-03 10:31:36.744 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1706030a1f230dc40014000002c00ad20023000003d90000000000000000000000000000000000000202098b001c1385000000ad095600191389000001be0970001a1384ffffff310001000001b60000000900000000000000000975000613850001000000700947000c13890001000000a7095c000d13840001000000b20000002e0000000600000002000001c500000018000701a8000001ce40201edf0f6f0858ffccfffffba90003000000090001000000000000000010b50000003500000f490000021c001f000000df000400000fcb002c000005a6001f000004c3000b0004000000010000000000000000000002000800000204530000274d
2023-06-03 10:31:36.745 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-06-03 10:31:36.835 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff0192000100cc001900190000002f006300040000012200000000000000010111031210120e1100cf00cb0d1e0d168f2e
2023-06-03 10:31:36.836 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-03 10:31:37.128 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a0001005e000a01110001006d011dfe80000afd7500b702c1fd64fffd13884803db004820c5800000006d0000011dfffffe800000000afffffe51ffffffedffffff38fffffd75000001db00000190fffffde900000583000100053f24
2023-06-03 10:31:37.129 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.686 seconds (success: True)
2023-06-03 10:31:42.443 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-03 10:31:42.748 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1706030a1f290de40014000002c40abd0023000003d20000000000000000000000000000000000000202099b0014138600000174095e0005139a0000002d097d000e1381000000210001000001c30000000e000000000000000009830006138600010000007b094f000c139a0001000000a1096c000e13810001000000c40000003e0000001600000005000001ce00000019000701a8000001cd40201f0a0f830856ffccfffffba90003000000090001000000000000000010b50000003500000f490000021c001f000000df000400000fcb002c000005a6001f000004c3000b00040000000100000000000000000000020008000002046800008eab
2023-06-03 10:31:42.749 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-06-03 10:31:42.842 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff0192000100cc001900190000002f006300040000012200000000000000010111031210120e1100cf00cb0d1a0d12cf2c
2023-06-03 10:31:42.845 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-03 10:31:43.136 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a0001005e000a011100010136ff79ff630011011d0355fea7fe27000f138b4803db004820c58000000136ffffff79ffffff6300000011ffffffba000000f50000006e0000011d00000168fffffe92fffffeae00000429000100057247
2023-06-03 10:31:43.154 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.712 seconds (success: True)
2023-06-03 10:31:48.450 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-03 10:31:48.787 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1706030a1f2f0de20014000002c60adc0023000003c90000000000000000000000000000000000000202098f000e138d000000f20951000a13820000007f0977000e138f0000005b0001000001d00000001f000000000000000009770005138d0001000000370944000c13820001000000b20962000c138f0001000000b00000000f0000000e00000014000001aa00000007000601a8000001cd40201eed0f730855ffccfffffbab0003000000090001000000000000000010b50000003500000f490000021c001f000000df000400000fcb002c000005a6001f000004c3000b00040000000100000000000000000000020008000002046f00000a57
2023-06-03 10:31:48.804 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-06-03 10:31:49.833 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70390880018fc7c in time (1s).
2023-06-03 10:31:49.836 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c - retry #1/10
2023-06-03 10:31:50.901 DEBUG (MainThread) [goodwe.protocol] Failed to receive response to f70390880018fc7c in time (1s).
2023-06-03 10:31:50.910 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c - retry #2/10
2023-06-03 10:31:51.079 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff0192000100cc001900190000002f006300040000012200000000000000010111031110120e1100cf00cc0d180d105f2e
2023-06-03 10:31:51.091 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-03 10:31:51.301 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a0001005e000a01110001ffef00300002002001adffce00c30002002d13864803db004820c580ffffffef000000300000000200000020000000f70000005c00000059000001adfffffeff000000df000000cd000002af000100051d1a
2023-06-03 10:31:51.304 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 2.860 seconds (success: True)
2023-06-03 10:31:56.443 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-03 10:31:56.750 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa1706030a1f370e000014000002cc0ab40024000003da000000000000000000000000000000000000020209930004138500000037095d000d1384000000e2097b00141383000000be0001000001db000000150000000000000000 [... cut ...]
2023-06-03 10:33:06.838 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-03 10:33:07.128 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a0001005e000a01110001fcf4fd3ffd05f738fb72fca5fcbafc79fc9e13864803db004820c800fffffcf4fffffd3ffffffd05fffff738fffffe36fffffe66fffffed8fffffb72fffffc70fffffcb2fffffcb1fffff5d200010005498d
2023-06-03 10:33:07.129 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.685 seconds (success: True)
'''

filipkrystl, 2023-06-16:
Same for me, only difference is that power is always set to 50% no matter what my "Eco mode power" is set.

lampra1, 2023-12-24:
Probably a firmware issue. Mine works fine.
 Firmware: 04029-10-S11 / 02041-25-S00

ahammas, 2024-01-28:
I'm experiencing the same thing however I can't compare the FW version that you state. My inverter (10kn-ET) reports fw 101025. Any idea if this issue is FW related in my case?

bakerboy908, 2024-10-23:
Closing this issue, its still not working correctly for me, but a work around im using is to set the battery charge current.

If you decide to do this, you need to make sure your dont set it higher then what the bms is calling for, or you can damage the cells or even cause a fire.

closed 2024-10-23 by bakerboy908