mletenay/home-assistant-goodwe-inverter #195: Firmware version (ARM) is wrong

smileyman-1, 2023-06-04:
In device information dialog the (ARM) firmware version is wrong: (Digits are swapped)

[image]

correct is (from PV-Master / SolarGo-App and firmware file): 17.17.53 
-> the ARM version is not 35, correct ist 53

nelgi, 2023-06-07:
a bit offtopic, but how you get new fw versions? mine is 14.14.10:

[image]

smileyman-1, 2023-06-10:
Open a ticket at Goodwe (and wait, wait, wait, wait ;-)) oder do it by yourself: https://www.photovoltaikforum.com/core/attachment/342066-bluetooth-firmware-update-string-storage-de-v002-pdf/

wesleyv010, 2023-07-01:
I have a new inverter last month. The firmware version is 19.19. 51.

Is this firmware more recent and better?

nelsnels, 2023-10-12:
> Open a ticket at Goodwe (and wait, wait, wait, wait ;-)) oder do it by yourself: https://www.photovoltaikforum.com/core/attachment/342066-bluetooth-firmware-update-string-storage-de-v002-pdf/

Hi @smileyman-1 , thank you for your work and for this document! Did you have the chance to update an hybrid inverter, such as series ET? I haven't found a way to use the SolarGo app to upgrade the firmware using the files in the Google drive. The app only suggest an automatic update, which is surprisingly stuck in ARM 7 in my case. Is there another way to perform this update?

As you suggested, I also open at ticket at Goodwe and I'm currently wainting, wainting, wainting...

smileyman-1, 2023-10-14:
Hi @nelsnels,
I only have one from the EM series and one from the SDT series.
With the SDT G2 I connect via Solar Go via WLAN and download the bin file from Google Drive beforehand. However, this must be renamed, otherwise the name is too long. I can select this file via 'Manual file selection' and flash it.
With the EM inverter, the whole thing only works via Bluetooth dongle. Unfortunately I do not have a Bluetooth dongle.

mletenay, 2024-05-25:
Closing the issue as obsolete/invalid.
There is actually no bug there, the value is taken as it is.
It's juts encoded in hex and accidentally 53 = 0x35 ;-)

closed 2024-05-25 by mletenay