mletenay/home-assistant-goodwe-inverter #196: Support for SEC Smart Energy Controller

thibaultmol, 2023-06-13:
SEC1000 (On Gird) / SEC1000S (Storage)
https://en.goodwe.com/sec1000-smart-energy-controller

thibaultmol, 2023-07-16:
I now have one installed since this week. So if you need any type of data, i could help

(so for context: it's meant for installations that have 3 inverters with a battery on each inverter. To basically balance the output of the on-grid output of the inverters when drawing from the batteries)

mletenay, 2023-12-21:
I do not know what kind of support you expect, but unless we get some manufacturer protocol documentation, nothing will happen.

closed 2023-12-21 by mletenay

thibaultmol, 2025-08-05:
I also haven't gotten any docs from goodwe. But for anyone finding this thread:
Here are the commands to turn on and off the export (to the grid) limiter.
```
Turn the limiter on: printf '\x00\x06\x01\x01\x40\x01\x00\x43' | nc [IP address of your SEC unit] 1234 | xxd -p
Turn the limiter off: printf '\x00\x06\x01\x01\x40\x02\x00\x44' | nc [IP address of your SEC unit] 1234 | xxd -p

```

Don't know how to change the limiter amount yet but atleast can turn it on off 
Only way I found to do it is via their application, which I used to reverse engineer the commands mentioned above, which I'll try to edit and update as I figure it out 
https://en.goodwe.com/Ftp/EN/Downloads/Software/GW_ProMate_V2.0.19.rar