mletenay/home-assistant-goodwe-inverter #199: goodwe.set_parameter service disappear

TomEkk33, 2023-06-27:
Hello,

Today after upgrade the HA to 2023.6.3 service goodwe.set_parameter disappear. 
Inverter in GW10k-ET, all entities seems to be OK.
Could you check what happened?

Home Assistant 2023.6.3
Supervisor 2023.06.2
HACS 1.32.1

It was described previously here.
https://github.com/mletenay/home-assistant-goodwe-inverter/issues/167

TomEkk33, 2023-06-27:
The only change in Home Assistant 2023.6.3 related to Goodwe is:
https://github.com/home-assistant/core/pull/95041

Does anyone else see missing service goodwe.set_parameter ?
The log shows error related to "homeassistant.components.automation.":
```
2023-06-27 14:45:47.110 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.689 seconds (success: True)
2023-06-27 14:45:47.457 DEBUG (MainThread) [custom_components.goodwe.select] Settin operation mode to general, power 2, max SoC 100
2023-06-27 14:45:47.461 DEBUG (MainThread) [goodwe.protocol] Sending: f706b79800003b07
2023-06-27 14:45:47.506 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b79800003b07
2023-06-27 14:45:47.509 DEBUG (MainThread) [goodwe.protocol] Sending: f710b99700020400010000a8f9
2023-06-27 14:45:47.754 DEBUG (MainThread) [goodwe.protocol] Received: aa55f710b9970002c1ee
2023-06-27 14:45:47.762 DEBUG (MainThread) [goodwe.protocol] Sending: f706b9ad0001e821
2023-06-27 14:45:48.005 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b9ad0001e821
2023-06-27 14:45:48.009 DEBUG (MainThread) [goodwe.protocol] Sending: f706b99627104210
2023-06-27 14:45:48.254 DEBUG (MainThread) [goodwe.protocol] Received: aa55f706b99627104210

2023-06-27 14:45:48.257 ERROR (MainThread) [homeassistant.components.automation.gwe_pc_exp_pwr_10kw_gen] GWE.PC 10kW, General: Error executing script. Service not found for call_service at pos 3: Unable to find service goodwe.set_parameter

2023-06-27 14:45:50.217 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2023-06-27 14:45:50.519 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa17061b0e2d3115a300220000075c029b0000000000000000000000000000000000000000000000000002099300081386ffffffd3099a00081384ffffffd209a200081386ffffffd50001ffffff7bffffff3e0000000000000000098d0002138600010000000c09a10002138400010000000f099e0002138600010000000a0000000a0000000600000006000000230000001a000101dd000001bd40201f1b0f890be7ffbffffff82900030000000d000100000000000000000522000000860000053b000000a400760000003500000000003700040000028900290000021700030006000000010000000000000000000003000800000207570000463e
2023-06-27 14:45:50.523 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2023-06-27 14:45:50.613 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff01020001015e000e0025000000590064000600000101000000000000000000000000000000000000000000000000d365
2023-06-27 14:45:50.617 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2023-06-27 14:45:50.905 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a00010042000a01110001ffbeffbdffc6ff4100b8fda0fd95fdcdfdab138647f2bc0045a8c000ffffffbeffffffbdffffffc6ffffff410000003a0000003d00000040000000b8ffffff8fffffff91ffffff94fffffeb3000100507d7e

```

Thanks.

TomEkk33, 2023-06-30:
Hello  @mletenay,
Today I noticed that **fast_charging** doesn't work, because it uses the same call service **goodwe.set_parameter**. 
This is more painfull, because it prevents automatic battery charging from the grid.

```
  - service: goodwe.set_parameter
    data:
      device_id: eeb524227071a5bd99578cb7ccc678ac
      parameter: fast_charging
      value: 1
    alias: Fast Charge = 1
```

TomEkk33, 2023-07-02:
Hello all,
I reverted to HA Core  2023.6.2, but  this problems still exists. I Enabled debugging of Goodwe - reinitialized device.
Then only this error is visible, but it is not related to service issue.
```
2023-07-02 02:12:31.693 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa170702020c1d0000000000000000026700000000000000000000000000000000000000000000000000000948000a13860000002e091e000a13860000001a0934000a13870000003f00010000008a00000062000000000000000009430002138600010000000a09270002138600010000000d0932000213870001000000040000000600000001000000040000001a0000000e000101970000014f40201e1b0f090bab00040000008c00020000000d0001000000000000000007de00000000000007e90000010f00030000003700000000005900010000037000000000027000000006000000000000000000000000000003080850000300c800007e64
2023-07-02 02:12:31.699 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 139, in update_eco_mode_power
    self._eco_mode_power = int(float(event.data.get("new_state").state))
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
2023-07-02 02:12:31.702 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 153, in update_eco_mode_soc
    self._eco_mode_soc = int(float(event.data.get("new_state").state))
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
```
Any hints?

joshuaaaaa, 2023-08-12:
I have the same problem :/

DvorakMartin1, 2023-09-19:
The Goodwe Inverter (experimental) service disappeared and with it the goodwe.set_parameter service after adding another DT inverter to the existing ET type.

DvorakMartin1, 2023-09-27:
If I have two inverters in SEMportal (in my case ET and DT) and it doesn't matter if they are together (they have the same PowerStatusSnMin or separately ... they have different PowerStatusSnMin), then after adding them everything works. After restarting HA, however, the parameter "GoodWe Inverter: Set inverter configuration parameter - EXPERIMENTAL is no longer available" and the supply to the grid cannot be controlled (ET version).
Availability with is only possible by removing and re-adding the second inverter. I expect the second inverter to only read the values (if it is powered by the panels) and that is functional, but its presence after the restart blocks the ET for controlling the supply to the grid.
Please repair.

mletenay, 2023-10-15:
I did not find clear explanation for set_parameter disappearance, but I did some potential fix.
Could you try to install master version and test it ?

DvorakMartin1, 2023-10-16:
I downloaded the master version and did a reboot and then another and
everything seems to be working.
The experimental version with two inverters did not disappear after a
restart.
The bug is fixed.
Home Assistant monitors my export size, and unauthorized exports above
the contractual limit are subject to heavy fines (the fine is 300 times
the price per kWh).
With a permanently activated limit, Goodwe has a high idle consumption.
Your function takes care of that. I have an ET and a DT inverter that
work indefinitely and I monitor the export size (allowed 9.9kW and
installed 15kW). If the export is >9900W, I activate the limit with a
setting of 9900W and deactivate it again after sunset. I have to rely on
this feature.
Many thanks, I am very pleased with it. 

Question: should I keep the master version forever or what about the
v0.9.9.11 version that is now being offered to me as an upgrade? 

Martin Dvorak
(Czech Republic)

 

Links:
------
[1]
https://github.com/mletenay/home-assistant-goodwe-inverter/issues/199#issuecomment-1763502803
[2]
https://github.com/

mletenay, 2023-10-25:
Closing as fixed in v0.9.9.12

closed 2023-10-25 by mletenay