mletenay/home-assistant-goodwe-inverter #200: Error decoding firmware version E1.00

extermi111, 2023-06-27:
```Error decoding firmware version E1.00.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/goodwe/es.py", line 197, in read_device_info
    self.dsp1_version = int(self.firmware[0:2])
                        ^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'E1'```

Name: Grid-Tied PV Inverter
Type: GW10K-DT

mletenay, 2023-10-15:
The problem is most likely not in decoding, butin inverter identification.
You have DT inverter, but the ES inverter code is being executed.
Does this problem  still exist ?

mletenay, 2023-10-26:
Just for refence, most likely the same problem as here - https://github.com/home-assistant/core/issues/95767

closed 2023-12-21 by mletenay