mletenay/home-assistant-goodwe-inverter #201: Get Eco mode value

extermi111, 2023-06-27:
```
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 153, in update_eco_mode_soc
    self._eco_mode_soc = int(float(event.data.get("new_state").state))
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
```

```
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 139, in update_eco_mode_power
    self._eco_mode_power = int(float(event.data.get("new_state").state))
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
```

Name: Grid-Tied PV Inverter
Type: GW10K-DT

TomEkk33, 2023-07-21:
Hello,
FYI it's a similar issue reported in https://github.com/mletenay/home-assistant-goodwe-inverter/issues/181, and still open

mletenay, 2023-10-15:
Fixed in acc4dbc126a102c4a0b94b8cef910e88b2d467ca , will get to next release (in couple of days)

closed 2023-10-15 by mletenay