mletenay/home-assistant-goodwe-inverter #203: Battery Level GOODWE 100%

liebefuroscar, 2023-07-03:
I have a DIY battery without communication in my goodwe EM 5048.
Although I have the charging voltage set to 55v, at 52v it already indicates that the battery percentage is 100%.
In configuration I have the SOC protection disabled, but I think that the Eco mode SoC sensor interrupts something and makes me unable to use 100% of my battery

TomEkk33, 2023-07-21:
Hello,
About ~52 V as 100% is the correct value for LFP battery with nominal voltage = 48 V. 
I have the same results with Pylon's LFP batteries. 
This is not an issue of Goodwe inverter.

midadami, 2023-08-03:
> Hello, About ~52 V as 100% is the correct value for LFP battery with nominal voltage = 48 V. I have the same results with Pylon's LFP batteries. This is not an issue of Goodwe inverter.

Be careful with this statement as I think it might lead to some confusion. Pylontech LFP batteries are usually 48V nominal but thats not the norm for DIY batteries. DIY batteries are 16S and therefore are 51.2V nominal so ~55.2V is usually used as the 100% voltage.

TomEkk33, 2023-08-03:
What I wrote strictly relates to Pylon's LFP long life batteries. 
I hoped it is clear.

Goodwe inverter has nothing to do with this I suppose.

liebefuroscar, 2023-08-03:
The report comes from HA's "Soc protect". Even though the battery is actually at % battery, Goodwe reports the % that sounds good for her.
It would be interesting to be able to "ignore" the SOC protect sensor in HA to allow charging or discharging the battery up to the voltage of interest in each case. (SOC protect is disabled in the inverter configuration) but it is the Home assistant sensor that limits in this case since it always has a numerical value, either 0 or 100. So even if it is disabled in the inverter configuration, Home assistant always reports a value. And that value is limiting.

TomEkk33, 2023-08-03:
HA is just an interface. All is inside the inverter and BMS.
Use SolarGo/PVMaster app and check the battery settings, related voltages, current, status 
available inside Goodwe inverter. 
Basing on this you can figure-out what happens in your system.

mletenay, 2023-10-15:
HA SoC is just a sensor and it can't limit anything. This is obviously not a HA issue.

closed 2023-10-15 by mletenay