mletenay/home-assistant-goodwe-inverter #204: Feature request

Kratax, 2023-07-20:
Hi,
Is it possible / would it be to set up (change) max battery charging power?
I think there would be a nice use case for that... thanks a lot!
Lubos

lampra1, 2023-08-01:
This is already avalable in Eco charge mode. The Eco mode power percentage refers to the rated power of the inverter. For example, if set at 10% for a 5Kw inverter the battery will be charging at 500w. Though, with this limit, depending on pv power, the battery could be chatging from the pv and/or the grid.

Kratax, 2023-08-13:
Thanks a lot for the advice. It is working but not fully. The issue is in a situation where there will be a need for usage of battery (i.e. consumption higher than solar production). I guess that as you have described the behavior, it will stay the same, i.e. charging with 1% power but not utilizing battery.

I was looking ideally for behavior where if there is enough energy, it is exported but if there is a need for battery power (ie consumption higher than production), the battery is used....

Obviously can be done by automation but the original request would work nicely as well without monitoring & running automations based on actual data...

lampra1, 2023-09-13:
> I was looking ideally for behavior where if there is enough energy, it is exported but if there is a need for battery power (ie consumption higher than production), the battery is used....

Set the eco charge/discharge mode to 0%. This emulates the GENERL MODE

closed 2023-12-21 by mletenay