mletenay/home-assistant-goodwe-inverter #207: SEMS portal update failure

alpzritz, 2023-08-15:
Hi , after configuration I notice I get data but inverter stops sending this data to SEMS portal so SEMS portal reports no updates. When I disable this integration SEMS portal starts reporting data as normal.

This has been issue since over a year , I did reach goodwe and asked them to update my firmware which they say it's already up to date. 

Any idea if I am doing something wrong here my inverter is GW15KAU-DT

JoshC1994, 2023-09-11:
possibly you are polling your inverter too often? 
[image]

alpine85, 2023-10-03:
I am having the same issue for also about 1 year, be great to solve this as HA can have gaps in lomg term data

JoshC1994, 2023-10-03:
> I am having the same issue for also about 1 year, be great to solve this as HA can have gaps in lomg term data

Have you reached out to Goodwe regarding the issue, What Inverter & what firmware version?

alpzritz, 2023-10-03:
I am using inverter model GW15KAU, I did contact goodie they can't help other than upgrading to latest firmware. Which they did already. Not guess there is some read and delete happening on memory so when HA reads this info it's also deleted and never reaches SEMS. Or inverter stops sending data to SEMS. I think I'll need to trace network to find out what is going on.

manitu1, 2023-10-04:
I am using GW29.9K-ET and I needed to change interval to 5 mins. If I set les than 5 mins HA works but SENS has no data.

Before I had GW10K-ET and I had 10 sec time interval with no issues.

smileyman-1, 2023-10-14:
I have a GW3648-EM. Here a query every second is no problem. SEMS works anyway.
With a GW10KT-DT, however, SEMS only works if the interval is >=180 seconds.

Maybe it would help to change the WIFI module to Multi UDP Connects.
However, I have not yet succeeded in controlling it.

joshuaaaaa, 2023-11-14:
Hi, I also have a problem with GW8K-DT and SEMS. After adding to HA, I have no data on SEMS. I will try to set an interval of 180 seconds.

mkubant, 2024-01-26:
Hi, anything new on this issue? I have the same problem with the GW15K-ET. I see that it is also solved with some workarounds here https://github.com/home-assistant/core/issues/101817

GIJOE-SA, 2024-03-18:
Also upgraded to the new G2 from my old ES and now have this issue. Tried to log a call with Goodwe but got this response: 

"Perhaps you can only use one app a time. We honestly can not provide a resolve when using a third party monitoring app. If the G2 wifi dongle is working on the inverter there is nothing else we can assist with."

Dwayneohara, 2024-05-18:
i feel a bit dumb, i have hacs installed but how do i install this experimental version of goodwe integration ?

JoshC1994, 2024-05-18:
You need to add the repository then you can search via hacs

Misenkooo, 2025-03-08:
Let me tell you something about this and possible dongle hardware issue. I don't it is like that. The reason is own experience. I've got inverter (GW10K-ET PLUS+) installed on 10.1.2025 with newest firmware 1230. I had no issues with simultaneous connection to SEMS and HA via goodwe  integration at the same time. Polling 5s no problems whatsoever! 2 months no issues at all.

But I had some issues with PV drop to 0 for few minutes in case of semi-cloudy environment. So goodwe support told me they've initiated downgrade to "special" firmware 1228... I have the issues since then. First I did not understand what's going on. I've tried everything, reconnect inverter, connect to different router, everything. Simply communication to SEMS was not working anymore. The only help was to restart inverter completelly - turn off, turn off. Pain. But it worked for few minutes only and then the same issue has occurred. Data were not sent to sems anymore. I was frustrated, sent tickets to goodwe, no response. Why the heck this is happening where I had no issuef for 2 monhts ?? Then I've god the idea to disable goodwe integration in HA and bang, the communication started working again!

This is definitely a firmware problem which obviously was fixed in the latest version. At least for this inverter model. I don't want to upgrade it to the latest version because of this dropouts I had before, I need to watch how it behaves. So it's pretty frustrating.