mletenay/home-assistant-goodwe-inverter #210: How to add inverter to HA using this integration?

vsisl, 2023-08-23:
Hello,

Thanks for developing this integration!

I just installed it from HACS and now I would like to add my inverter to HA using this integration. Note: I am using a fresh installation of HA and I did NOT add my GoodWe invertor using the official GoodWe integration before installing the experimental integration.

How do I add the inverter?

I know the IP address of the inverter and I also [successfully tested communication with the inverter using your script](https://github.com/mletenay/home-assistant-goodwe-inverter/tree/master#inverter-discovery-and-communication-testing).

Thank you for your help!

JoshC1994, 2023-09-11:
you need to add the inverter using the goodwe integration. Search for goodwe and input your IP address.
[image]

vsisl, 2023-10-01:
Hi @JoshC1994 ,

Thanks a lot for your reply. Indeed after installing `GoodWe Inverter (experimental)` from HACS one has to simply go to Settings -> Devices & services -> Add integration, and then search for `GoodWe Inverter`. What confused me is that the integration is called `GoodWe Inverter` instead of `GoodWe Inverter (experimental)`. I didn't notice the box icon indicating that this is a custom integration installed through HACS. 

[image]

closed 2023-10-01 by vsisl