mletenay/home-assistant-goodwe-inverter #211: Can't connect to GW15KN-DT with LAN module

wiljums, 2023-09-11:
I can't connect to my GW15KN-DT with LAN module (no wifi). With the standard Goodwe integration in HA I can. 
As the inverter has no wifi it also doesn't connect with the mobile apps SolarGo and PV Master.

Do I need a wifi connection with the inverter to be able to use this experimental integration via HACS?

wiljums, 2024-02-28:
Re-opening because I still have this issue. inverter_test.py gives 'no valid response'.

mletenay, 2024-02-28:
Provide logs from HA integration and HACS integration.
There is not reason why one should work and the other not, they are mostly the same.

wiljums, 2024-02-28:
Ok, after many tries it did connect. But the communication/connection remains fragile. Only a valid response in 1 of multiple requests (same as with the official integration).

closed 2024-02-28 by wiljums